platformio/platformio-core-installer #26: Test installer with different Linux distributives

ivankravets, 2020-03-16:
Please test with Python 2 and Python 3. Also, please provide in each new comment a `debug` output of each installation process:
```
python get-platformio.py --verbose
```

We need this for history.

- [ ] Ubuntu 18
- [ ] Ubuntu 16
- [ ] Debian 10
- [ ] Debian 9
- [ ] Fedora (latest)
- [ ] CentOS (latest)
- [ ] FreeBSD
- [ ] ArchLinux

ShahRustam, 2020-03-17:
### Ubuntu 18. 
### Python 2.7. 
### File: https://raw.githubusercontent.com/platformio/platformio-core-installer/develop/get-platformio.py 

root@ubuntu-2gb-hel1-1:~# python get-platformio.py --verbose
INFO: Installer version: 0.1.0
DEBUG: Invoke: get-platformio.py --verbose
DEBUG: Platform: Linux-4.15.0-74-generic-x86_64-with-Ubuntu-18.04-bionic
INFO: Python version: 2.7.17 (default, Nov  7 2019, 10:07:09)
[GCC 7.4.0]
INFO: Python path: /usr/bin/python
INFO: Creating a virtual environment at /root/.platformio/penv
DEBUG: Current environment PATH /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
DEBUG: Checking a Python candidate /usr/bin/python3
DEBUG: Found a compatible Python /usr/bin/python3
DEBUG: Checking a Python candidate /usr/bin/python
DEBUG: Found a compatible Python /usr/bin/python
DEBUG: Checking a Python candidate /usr/bin/python2
DEBUG: Found a compatible Python /usr/bin/python2
DEBUG: Using /usr/bin/python3 Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python3 -m venv /root/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv -p /usr/bin/python3 /root/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python3 /root/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv /root/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv /root/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python3 /root/.platformio/penv-tmp/virtualenv.pyz /root/.platformio/penv
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/root/.platformio/penv-tmp/virtualenv.pyz/__main__.py", line 169, in <module>
  File "/root/.platformio/penv-tmp/virtualenv.pyz/__main__.py", line 163, in run
  File "/root/.platformio/penv-tmp/virtualenv.pyz/virtualenv/__init__.py", line 3, in <module>
  File "/root/.platformio/penv-tmp/virtualenv.pyz/virtualenv/run/__init__.py", line 13, in <module>
  File "/root/.platformio/penv-tmp/virtualenv.pyz/virtualenv/run/plugin/creators.py", line 6, in <module>
  File "/root/.platformio/penv-tmp/virtualenv.pyz/virtualenv/create/via_global_ref/builtin/builtin_way.py", line 7, in <module>
  File "/root/.platformio/penv-tmp/virtualenv.pyz/virtualenv/create/creator.py", line 14, in <module>
  File "/root/.platformio/penv-tmp/virtualenv.pyz/virtualenv/discovery/cached_py_info.py", line 17, in <module>
  File "/root/.platformio/penv-tmp/virtualenv.pyz/virtualenv/discovery/py_info.py", line 16, in <module>
ModuleNotFoundError: No module named 'distutils.command'
DEBUG: Could not create virtualenv with downloaded script. Error: Command '['/usr/bin/python3', '/root/.platformio/penv-tmp/virtualenv.pyz', '/root/.platformio/penv']' returned non-zero exit status 1
DEBUG: Using /usr/bin/python Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python -m venv /root/.platformio/penv
/usr/bin/python: No module named venv
DEBUG: Creating virtual environment: /usr/bin/python -m virtualenv -p /usr/bin/python /root/.platformio/penv
/usr/bin/python: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python /root/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python -m virtualenv /root/.platformio/penv
/usr/bin/python: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv /root/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python /root/.platformio/penv-tmp/virtualenv.pyz /root/.platformio/penv
INFO: Updating Python package manager (PIP) in a virtual environment
DEBUG: Creating pip.conf file in /root/.platformio/penv
DEBUG: Downloading 'get-pip.py' installer...
DEBUG: Installing pip
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
INFO: PIP has been successfully updated!
INFO: Virtual environment has been successfully created!
INFO: Installing PlatformIO Core
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
INFO: PlatformIO Core has been successfully installed!

ShahRustam, 2020-03-17:
Ubuntu 18.
Python 2.7.
File: https://raw.githubusercontent.com/ShahRustam/platformio-core/develop/scripts/get-platformio.py
✅

rs@ubuntu-2gb-hel1-1:~$ python get-platformio.py --verbose
This installer script is deprecated and will be removed in the next release. Please use https://raw.githubusercontent.com/platformio/platformio-core-installer/develop/get-platformio.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  761k  100  761k    0     0  2780k      0 --:--:-- --:--:-- --:--:-- 2780k
INFO: Installer version: 0.1.0
DEBUG: Invoke: /tmp/tmpqWq65U --verbose
DEBUG: Platform: Linux-4.15.0-74-generic-x86_64-with-Ubuntu-18.04-bionic
INFO: Python version: 2.7.17 (default, Nov  7 2019, 10:07:09)
[GCC 7.4.0]
INFO: Python path: /usr/bin/python
INFO: Creating a virtual environment at /home/rs/.platformio/penv
DEBUG: Current environment PATH /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
DEBUG: Checking a Python candidate /usr/bin/python3
DEBUG: Found a compatible Python /usr/bin/python3
DEBUG: Checking a Python candidate /usr/bin/python
DEBUG: Found a compatible Python /usr/bin/python
DEBUG: Checking a Python candidate /usr/bin/python2
DEBUG: Found a compatible Python /usr/bin/python2
DEBUG: Using /usr/bin/python3 Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python3 -m venv /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv -p /usr/bin/python3 /home/rs/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python3 /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv /home/rs/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv /home/rs/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python3 /home/rs/.platformio/penv-tmp/virtualenv.pyz /home/rs/.platformio/penv
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/__main__.py", line 169, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/__main__.py", line 163, in run
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/__init__.py", line 3, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/run/__init__.py", line 13, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/run/plugin/creators.py", line 6, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/create/via_global_ref/builtin/builtin_way.py", line 7, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/create/creator.py", line 14, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/discovery/cached_py_info.py", line 17, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/discovery/py_info.py", line 16, in <module>
ModuleNotFoundError: No module named 'distutils.command'
DEBUG: Could not create virtualenv with downloaded script. Error: Command '['/usr/bin/python3', '/home/rs/.platformio/penv-tmp/virtualenv.pyz', '/home/rs/.platformio/penv']' returned non-zero exit status 1
DEBUG: Using /usr/bin/python Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python -m venv /home/rs/.platformio/penv
/usr/bin/python: No module named venv
DEBUG: Creating virtual environment: /usr/bin/python -m virtualenv -p /usr/bin/python /home/rs/.platformio/penv
/usr/bin/python: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python -m virtualenv /home/rs/.platformio/penv
/usr/bin/python: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv /home/rs/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python /home/rs/.platformio/penv-tmp/virtualenv.pyz /home/rs/.platformio/penv
INFO: Updating Python package manager (PIP) in a virtual environment
DEBUG: Creating pip.conf file in /home/rs/.platformio/penv
DEBUG: Downloading 'get-pip.py' installer...
DEBUG: Installing pip
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
INFO: PIP has been successfully updated!
INFO: Virtual environment has been successfully created!
INFO: Installing PlatformIO Core
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
INFO: PlatformIO Core has been successfully installed!

ShahRustam, 2020-03-17:
### Ubuntu 18.
### Python 3.6 - with installed python3-distutils
### File: https://raw.githubusercontent.com/platformio/platformio-core-installer/develop/get-platformio.py
✅

rs@ubuntu-2gb-hel1-1:~$ python3 get-platformio.py --verbose
INFO: Installer version: 0.1.0
DEBUG: Invoke: get-platformio.py --verbose
DEBUG: Platform: Linux-4.15.0-74-generic-x86_64-with-Ubuntu-18.04-bionic
INFO: Python version: 3.6.9 (default, Nov  7 2019, 10:44:02)
[GCC 8.3.0]
INFO: Python path: /usr/bin/python3
INFO: Creating a virtual environment at /home/rs/.platformio/penv
DEBUG: Current environment PATH /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
DEBUG: Checking a Python candidate /usr/bin/python3
DEBUG: Found a compatible Python /usr/bin/python3
DEBUG: Checking a Python candidate /usr/bin/python
DEBUG: Found a compatible Python /usr/bin/python
DEBUG: Checking a Python candidate /usr/bin/python2
DEBUG: Found a compatible Python /usr/bin/python2
DEBUG: Using /usr/bin/python3 Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python3 -m venv /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv -p /usr/bin/python3 /home/rs/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python3 /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv /home/rs/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv /home/rs/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory: 'virtualenv': 'virtualenv'
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python3 /home/rs/.platformio/penv-tmp/virtualenv.pyz /home/rs/.platformio/penv
INFO: Updating Python package manager (PIP) in a virtual environment
DEBUG: Creating pip.conf file in /home/rs/.platformio/penv
DEBUG: Downloading 'get-pip.py' installer...
DEBUG: Installing pip
INFO: PIP has been successfully updated!
INFO: Virtual environment has been successfully created!
INFO: Installing PlatformIO Core
INFO: PlatformIO Core has been successfully installed!

ShahRustam, 2020-03-17:
### Ubuntu 18.
### Python 3.6 - with installed python3-distutils
### File: https://raw.githubusercontent.com/ShahRustam/platformio-core/develop/scripts/get-platformio.py
✅

rs@ubuntu-2gb-hel1-1:~$ python3 get-platformio.py --verbose
This installer script is deprecated and will be removed in the next release. Please use https://raw.githubusercontent.com/platformio/platformio-core-installer/develop/get-platformio.py
INFO: Installer version: 0.1.0
DEBUG: Invoke: /tmp/tmpldv88guf --verbose
DEBUG: Platform: Linux-4.15.0-74-generic-x86_64-with-Ubuntu-18.04-bionic
INFO: Python version: 3.6.9 (default, Nov  7 2019, 10:44:02)
[GCC 8.3.0]
INFO: Python path: /usr/bin/python3
INFO: Creating a virtual environment at /home/rs/.platformio/penv
DEBUG: Current environment PATH /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
DEBUG: Checking a Python candidate /usr/bin/python3
DEBUG: Found a compatible Python /usr/bin/python3
DEBUG: Checking a Python candidate /usr/bin/python
DEBUG: Found a compatible Python /usr/bin/python
DEBUG: Checking a Python candidate /usr/bin/python2
DEBUG: Found a compatible Python /usr/bin/python2
DEBUG: Using /usr/bin/python3 Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python3 -m venv /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv -p /usr/bin/python3 /home/rs/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python3 /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv /home/rs/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv /home/rs/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory: 'virtualenv': 'virtualenv'
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python3 /home/rs/.platformio/penv-tmp/virtualenv.pyz /home/rs/.platformio/penv
INFO: Updating Python package manager (PIP) in a virtual environment
DEBUG: Creating pip.conf file in /home/rs/.platformio/penv
DEBUG: Downloading 'get-pip.py' installer...
DEBUG: Installing pip
INFO: PIP has been successfully updated!
INFO: Virtual environment has been successfully created!
INFO: Installing PlatformIO Core
INFO: PlatformIO Core has been successfully installed!

ShahRustam, 2020-03-17:
### Ubuntu 16
### Python 3.5.2

### File: https://raw.githubusercontent.com/platformio/platformio-core-installer/develop/get-platformio.py

rs@ubuntu-2gb-hel1-2python3 get-platformio.py --verbose
Traceback (most recent call last):
  File "get-platformio.py", line 52, in <module>
    main()
  File "get-platformio.py", line 45, in main
    bootstrap()
  File "get-platformio.py", line 31, in bootstrap
    pioinstaller.__main__.main()
  File "/tmp/tmppollaswi/pioinstaller.zip/pioinstaller/__main__.py", line 82, in main
  File "/tmp/tmppollaswi/pioinstaller.zip/click/core.py", line 764, in __call__
  File "/tmp/tmppollaswi/pioinstaller.zip/click/core.py", line 696, in main
  File "/tmp/tmppollaswi/pioinstaller.zip/click/_unicodefun.py", line 124, in _verify_python3_env
RuntimeError: Click will abort further execution because Python 3 was configured to use ASCII as encoding for the environment. Consult https://click.palletsprojects.com/en/7.x/python3/ for mitigation steps.

This system supports the C.UTF-8 locale which is recommended.
You might be able to resolve your issue by exporting the
following environment variables:

    export LC_ALL=C.UTF-8
    export LANG=C.UTF-8

Click discovered that you exported a UTF-8 locale
but the locale system could not pick up from it because
it does not exist.  The exported locale is "en_US.UTF-8" but it
is not supported

### File: https://raw.githubusercontent.com/platformio/platformio-core-installer/Update-template.py/get-platformio.py

rs@ubuntu-2gb-hel1-2:~$ python3 get-platformio.py --verbose
INFO: Installer version: 0.1.0
DEBUG: Invoke: get-platformio.py --verbose
DEBUG: Platform: Linux-4.4.0-171-generic-x86_64-with-Ubuntu-16.04-xenial
INFO: Python version: 3.5.2 (default, Oct  8 2019, 13:06:37)
[GCC 5.4.0 20160609]
INFO: Python path: /usr/bin/python3
INFO: Creating a virtual environment at /home/rs/.platformio/penv
DEBUG: Current environment PATH /home/rs/bin:/home/rs/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
DEBUG: Checking a Python candidate /usr/bin/python3
DEBUG: Found a compatible Python /usr/bin/python3
DEBUG: Using /usr/bin/python3 Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python3 -m venv /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv -p /usr/bin/python3 /home/rs/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python3 /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv /home/rs/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv /home/rs/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory: 'virtualenv'
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python3 /home/rs/.platformio/penv-tmp/virtualenv.pyz /home/rs/.platformio/penv
INFO: Updating Python package manager (PIP) in a virtual environment
DEBUG: Creating pip.conf file in /home/rs/.platformio/penv
DEBUG: Downloading 'get-pip.py' installer...
DEBUG: Installing pip
INFO: PIP has been successfully updated!
INFO: Virtual environment has been successfully created!
INFO: Installing PlatformIO Core
INFO: PlatformIO Core has been successfully installed!

ShahRustam, 2020-03-17:
### Debian 10
### Python 3.7.3
### File: https://raw.githubusercontent.com/platformio/platformio-core-installer/develop/get-platformio.py

rs@debian-2gb-hel1-1:~$ python get-platformio.py --verbose
INFO: Installer version: 0.1.0
DEBUG: Invoke: get-platformio.py --verbose
DEBUG: Platform: Linux-4.19.0-6-amd64-x86_64-with-debian-10.2
INFO: Python version: 2.7.16 (default, Oct 10 2019, 22:02:15)
[GCC 8.3.0]
INFO: Python path: /usr/bin/python
INFO: Creating a virtual environment at /home/rs/.platformio/penv
DEBUG: Current environment PATH /usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
DEBUG: Checking a Python candidate /usr/bin/python3
DEBUG: Found a compatible Python /usr/bin/python3
DEBUG: Checking a Python candidate /bin/python3
DEBUG: Found a compatible Python /bin/python3
DEBUG: Checking a Python candidate /usr/bin/python
DEBUG: Found a compatible Python /usr/bin/python
DEBUG: Checking a Python candidate /bin/python
DEBUG: Found a compatible Python /bin/python
DEBUG: Checking a Python candidate /usr/bin/python2
DEBUG: Found a compatible Python /usr/bin/python2
DEBUG: Checking a Python candidate /bin/python2
DEBUG: Found a compatible Python /bin/python2
DEBUG: Using /usr/bin/python3 Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python3 -m venv /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv -p /usr/bin/python3 /home/rs/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python3 /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv /home/rs/.platformio/penv
/usr/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv /home/rs/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python3 /home/rs/.platformio/penv-tmp/virtualenv.pyz /home/rs/.platformio/penv
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/__main__.py", line 169, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/__main__.py", line 163, in run
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/__init__.py", line 3, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/run/__init__.py", line 13, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/run/plugin/creators.py", line 6, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/create/via_global_ref/builtin/builtin_way.py", line 7, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/create/creator.py", line 14, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/discovery/cached_py_info.py", line 17, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/discovery/py_info.py", line 16, in <module>
ModuleNotFoundError: No module named 'distutils.command'
DEBUG: Could not create virtualenv with downloaded script. Error: Command '['/usr/bin/python3', '/home/rs/.platformio/penv-tmp/virtualenv.pyz', '/home/rs/.platformio/penv']' returned non-zero exit status 1
DEBUG: Using /bin/python3 Python for virtual environment.
DEBUG: Creating virtual environment: /bin/python3 -m venv /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /bin/python3 -m virtualenv -p /bin/python3 /home/rs/.platformio/penv
/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv -p /bin/python3 /home/rs/.platformio/penv
DEBUG: Creating virtual environment: /bin/python3 -m virtualenv /home/rs/.platformio/penv
/bin/python3: No module named virtualenv
DEBUG: Creating virtual environment: virtualenv /home/rs/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /bin/python3 /home/rs/.platformio/penv-tmp/virtualenv.pyz /home/rs/.platformio/penv
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/__main__.py", line 169, in <module>
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/__main__.py", line 163, in run
[... 6 lines cut ...]
  File "/home/rs/.platformio/penv-tmp/virtualenv.pyz/virtualenv/discovery/py_info.py", line 16, in <module>
ModuleNotFoundError: No module named 'distutils.command'
DEBUG: Could not create virtualenv with downloaded script. Error: Command '['/bin/python3', '/home/rs/.platformio/penv-tmp/virtualenv.pyz', '/home/rs/.platformio/penv']' returned non-zero exit status 1
DEBUG: Using /usr/bin/python Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python -m venv /home/rs/.platformio/penv
/usr/bin/python: No module named venv
[... cut ...]
INFO: PIP has been successfully updated!
INFO: Virtual environment has been successfully created!
INFO: Installing PlatformIO Core
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
INFO: PlatformIO Core has been successfully installed!

ShahRustam, 2020-03-18:
### UBUNTU 18
_________________
Python 2 not exists
Python 3 - 3.6.9

After installed distutils - python3 create virtualenv with remote virtualenv package. All ok for native and wrapper script.

After installed python2 (apt-get install python) - create virtualenv with remote virtualenv package with ignore option for python3. All ok for native and wrapper script.

### UBUNTU 16
_________________
Python 2 not exists
Python 3  - 3.5.2

Error with python3 - click module: requires fixing locales.
export LC_ALL=C.UTF-8
export LANG=C.UTF-8

After  fixing locales - python3 create virtualenv with remote virtualenv package. All ok for native and wrapper script.

For python 2 install need apt-get update.
After installed python2 (apt-get install python) - create virtualenv with remote virtualenv package with ignore option for python3. All ok for native and wrapper script.

### FEDORA 31
_________________
Python 2 not exists
Python 3  - 3.7.5

python3 create virtualenv with local venv package. All ok for native and wrapper script.

After installed python2 (dnf install python2) - create virtualenv with remote virtualenv package with ignore option for python3. All ok for native and wrapper script.

### DEBIAN 10
_________________
Python 2 - 2.7.16
Python 3  - 3.7.3

After installed distutils - python3 create virtualenv with remote virtualenv package. All ok for native and wrapper script.

python2 create virtualenv with remote virtualenv package with ignore option for python3. All ok for native and wrapper script.

### DEBIAN 9
_________________
Python 2 - 2.7.13
Python 3  - 3.5.3

Error with python3 - click module: requires fixing locales.
export LC_ALL=C.UTF-8
export LANG=C.UTF-8

After  fixing locales - python3 create virtualenv with remote virtualenv package. All ok for native and wrapper script.

python2 create virtualenv with remote virtualenv package with ignore option for python3. All ok for native and wrapper script.

### CENT OS 8
_________________
Python 2 not exists
Python 3  - 3.6.8

python3 create virtualenv with local venv package. All ok for native and wrapper script.

After installed python2 (yum install python2) - create virtualenv with remote virtualenv package with ignore option for python3. All ok for native and wrapper script.

closed 2020-09-11 by ivankravets