platformio/platformio-core-installer #41: Unable to install platformio cli following official guide

akamensky, 2020-03-21:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Macos 10.15.3

**PlatformIO Version** (`platformio --version`): N/A

### Description of problem

```
$ python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)"
Installer version: 0.1.0
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
Platform: Darwin-19.3.0-x86_64-i386-64bit
Python version: 3.7.6 (default, Dec 30 2019, 19:38:26)
[Clang 11.0.0 (clang-1100.0.33.16)]
Python path: /usr/local/opt/python/bin/python3.7
Creating a virtual environment at /Users/alexey.kamenskiy/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory

Error: Could not install PlatformIO Core: Command '['/Users/alexey.kamenskiy/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "<string>", line 105, in <module>
  File "<string>", line 101, in main
  File "/usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/local/opt/python/bin/python3.7', '/var/folders/j6/vhm757px2zz3jgvxy1w3gfzr0000gn/T/tmpbhpyiuz0']' returned non-zero exit status 1.
```

#### Steps to Reproduce

1. Above
2.
3.

### Actual Results
Above

### Expected Results
Well, expected it to install...

ivankravets, 2020-03-22:
It looks like a directory (CWD) is removed. Could you try `cd ~/Downloads` and re-run installer?

akamensky, 2020-03-22:
@ivankravets You are right. Looks like I've tried a bunch of times from different tabs, when in all those tabs I was in a directory I previously deleted 🤦‍♂

closed 2020-03-22 by akamensky