platformio/platformio-core-installer #50: Installation Manager

werners7, 2020-06-09:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.46.0-insider
  PIO IDE: v2.0.0-beta.1
  System: Windows_NT, 10.0.19640, x64

  %23 Exception
  ```
  TypeError: Cannot read property 'core_dir' of null
	at Object.c [as getCoreDir] (c:\Users\werne\.vscode-insiders\extensions\platformio.platformio-ide-2.0.0-beta.1\node_modules\platformio-node-helpers\dist\index.js:1:5569)
	at e.callInstallerScript (c:\Users\werne\.vscode-insiders\extensions\platformio.platformio-ide-2.0.0-beta.1\node_modules\platformio-node-helpers\dist\index.js:1:33184)
	at t.value (c:\Users\werne\.vscode-insiders\extensions\platformio.platformio-ide-2.0.0-beta.1\node_modules\platformio-node-helpers\dist\index.js:1:28807)
	at processTicksAndRejections (internal/process/task_queues.js:85:5)
	at async Promise.all (index 0)
	at async c:\Users\werne\.vscode-insiders\extensions\platformio.platformio-ide-2.0.0-beta.1\dist\extension.js:1:5927
  ```

ivankravets, 2020-06-09:
Could you open VSCode > Help > Toggle developer tools and provide an output from a console?

rejnok93, 2020-07-02:
Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues -> Installer version: 0.3.2 Platform: Darwin-15.6.0 Python version: 2.7.10 (default, Oct 23 2015, 19:19:21) [GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.59.5)] Python path: /usr/bin/python Creating a virtual environment at /Users/rejnok/.platformio/penv
```
at f (/Users/rejnok/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:4975)
at ChildProcess.c (/Users/rejnok/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:4871)
at ChildProcess.emit (events.js:182:13)
at maybeClose (internal/child_process.js:962:16)
at Socket.stream.socket.on (internal/child_process.js:381:11)
at Socket.emit (events.js:182:13)
at Pipe._handle.close (net.js:606:12)
```
Atom 1.46.0 x64 on MacOS 10.11 (MacBook Pro Mid2009)
PlatformIO (version don't know, because many buttons and PIOhome button not working!)

RESULT: build not work (left sidebar button and inside PlatformIO menu do nothing), but if I press hotkey in console appears:
_Unable to execute: platformio
Make sure cmd:'platformio' and cwd:'/Users/rejnok/Dropbox/software/arduino/esp8266/scuremote' exists and have correct access permissions.
Binaries are found in these folders: /usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin_
Also upload not work (button, neither menu neither hotkey), and home button and item menu do nothing.

Before two weeks, I posted similar issue https://github.com/platformio/platformio-atom-ide/issues/2408 (but now the old error never appearing)

closed 2020-09-11 by ivankravets