platformio/platformio-core-installer #51: InstallationManager

rejnok93, 2020-07-02:
Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues -> Installer version: 0.3.2 Platform: Darwin-15.6.0 Python version: 2.7.10 (default, Oct 23 2015, 19:19:21) [GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.59.5)] Python path: /usr/bin/python Creating a virtual environment at /Users/rejnok/.platformio/penv

at f (/Users/rejnok/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:4975)
at ChildProcess.c (/Users/rejnok/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:4871)
at ChildProcess.emit (events.js:182:13)
at maybeClose (internal/child_process.js:962:16)
at Socket.stream.socket.on (internal/child_process.js:381:11)
at Socket.emit (events.js:182:13)
at Pipe._handle.close (net.js:606:12)
Atom 1.46.0 x64 on MacOS 10.11 (MacBook Pro Mid2009)
PlatformIO (version don't know, because many buttons and PIOhome button not working!)

RESULT: build not work (left sidebar button and inside PlatformIO menu do nothing), but if I press hotkey in console appears:
Unable to execute: platformio
Make sure cmd:'platformio' and cwd:'/Users/rejnok/Dropbox/software/arduino/esp8266/scuremote' exists and have correct access permissions.
Binaries are found in these folders: /usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin
Also upload not work (button, neither menu neither hotkey), and home button and item menu do nothing.

Before two weeks, I posted similar issue platformio/platformio-atom-ide#2408 (but now the old error never appearing)

ivankravets, 2020-07-02:
Could you try to install the latest Python 3 as instructed here https://docs.platformio.org/en/latest/faq.html#install-python-interpreter and restart Atom? Does it work now?

rejnok93, 2020-07-02:
Yes, it helped, now everythink is ok. Thanks.

closed 2020-07-02 by rejnok93

ivankravets, 2020-07-02:
Thanks for the report!