platformio/platformio-core-installer #55: get-platformio.py hangs after installing packages

ruspl-afed, 2020-08-12:
### Configuration

**Operating system**:
Windows 10

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 4.3.4 

### Description of problem
`get-platformio.py` does not complete the installation, it hangs after `Successfully installed` message
```cmd
Installer version: 0.3.3
Platform: Windows-10
Python version: 3.8.5 (tags/v3.8.5:580fbb0, Jul 20 2020, 15:57:54) [MSC v.1924 64 bit (AMD64)]
Python path: C:\soft\Python38\python.exe
Creating a virtual environment at <current_user_home_dir>\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Processing <current_user_home_dir>\appdata\local\pip\cache\wheels\22\76\4e\8451b93142686519520b707538b8ced937c1b7a3e79f369ef5\platformio-4.3.4-py3-none-any.whl
Collecting click!=7.1,!=7.1.1,<8,>=5
  Using cached click-7.1.2-py2.py3-none-any.whl (82 kB)
Collecting colorama
  Using cached colorama-0.4.3-py2.py3-none-any.whl (15 kB)
Collecting tabulate<1,>=0.8.3
  Using cached tabulate-0.8.7-py3-none-any.whl (24 kB)
Collecting pyelftools<1,>=0.25
  Using cached pyelftools-0.26-py2.py3-none-any.whl (136 kB)
Collecting semantic-version<3,>=2.8.1
  Using cached semantic_version-2.8.5-py2.py3-none-any.whl (15 kB)
Collecting pyserial!=3.3,<4,>=3
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting marshmallow>=2
  Using cached marshmallow-3.7.1-py2.py3-none-any.whl (45 kB)
Collecting requests<3,>=2.4.0
  Using cached requests-2.24.0-py2.py3-none-any.whl (61 kB)
Collecting bottle<0.13
  Using cached bottle-0.12.18-py3-none-any.whl (89 kB)
Collecting certifi>=2017.4.17
  Using cached certifi-2020.6.20-py2.py3-none-any.whl (156 kB)
Collecting chardet<4,>=3.0.2
  Using cached chardet-3.0.4-py2.py3-none-any.whl (133 kB)
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1
  Using cached urllib3-1.25.10-py2.py3-none-any.whl (127 kB)
Collecting idna<3,>=2.5
  Using cached idna-2.10-py2.py3-none-any.whl (58 kB)
Installing collected packages: click, colorama, tabulate, pyelftools, semantic-version, pyserial, marshmallow, certifi, chardet, urllib3, idna, requests, bottle, platformio
Successfully installed bottle-0.12.18 certifi-2020.6.20 chardet-3.0.4 click-7.1.2 colorama-0.4.3 idna-2.10 marshmallow-3.7.1 platformio-4.3.4 pyelftools-0.26 pyserial-3.4 requests-2.24.0 semantic-version-2.8.5 tabulate-0.8.7 urllib3-1.25.10
```

Removing `<current_user_home_dir>\.platformio` folder does not help.

ivankravets, 2020-08-12:
Thanks for the report! It seems it hangs at https://github.com/platformio/platformio-core-installer/blob/develop/pioinstaller/core.py#L100 , see also https://github.com/platformio/platformio-core-installer/blob/develop/pioinstaller/home.py#L49

It means that default pre-built packages are not compatible with your Python 3 installation.

How did you install Python 3?

ruspl-afed, 2020-08-12:
Thank you for reply! Where can I find a log to see and publish the errors?

> How did you install Python 3?

Well, I strated without any python. The installer tries to bring python37 but this also does not work for me.
Then I downloaded "Windows x86-64 executable installer" (`python-3.8.5-amd64.exe`) from https://www.python.org/downloads/release/python-385/ and installed the python using wizard. Nothing extra special, may be I unchecked some extra debug components that I don't need.

ruspl-afed, 2020-08-12:
And of cause I removing `~\.platformio` before every attempt to install PlatformIO

ivankravets, 2020-08-12:
If just tried to reproduce this issue in VM Win 10 using the same scenario:
[image]

1. Do you have other Pythons in the system?
2. Do you have 3rd party antivirus tools? They can block our installer.

Thanks!

ivankravets, 2020-08-12:
I forgot you need to wait a little bit after "Successfully installed...". Installer continues installtion and downloads PIO Home packages. It seems that we should add some info here. Could you wait a little bit?

ruspl-afed, 2020-08-12:
> Do you have other Pythons in the system?

Nope. Only installed Python 3.8 and its launcher.
```cmd
_myhome_>where python
C:\soft\Python38\python.exe
_myhome_\AppData\Local\Microsoft\WindowsApps\python.exe
```
> Do you have 3rd party antivirus tools? They can block our installer.

Nope, nothing special

> Could you wait a little bit?

One time I left it for about an hour. But it does not demonstrate any network activity 

Do you have an option to enable logging?

ivankravets, 2020-08-12:
>  _myhome_\AppData\Local\Microsoft\WindowsApps\python.exe

It looks like a broken Python from MS marketplace

> Do you have an option to enable logging?

When it hangs, just close that window and do not remove `.platformio` folder. Then run the next commands:

```
C:/Users/xxxx/.platformio/penv/Scripts/activate
pio home
```

Now, you should see some errors.

ruspl-afed, 2020-08-12:
The `pio home` command also hangs :)

ruspl-afed, 2020-08-12:
OK, I removed all the pythons , removed `~\.platformio` and tried to install `platformio-ide-2.0.0-beta.5.vsix` for VSCode
It is still installing...

ivankravets, 2020-08-12:
> The pio home command also hangs :)

Could you provide a screenshot from CLI?

ivankravets, 2020-08-12:
Do you use 3rd party antivirus tools?

ruspl-afed, 2020-08-12:
> > The pio home command also hangs :)
> 
> Could you provide a screenshot from CLI?

Unfortunately not, as it was before removing all the pythons. Now I have 
`(penv) C:\Users\xxxx\.platformio\penv\Scripts>pip home
ERROR: unknown command "home"`

ivankravets, 2020-08-12:
`pio`, not `pip`.

ruspl-afed, 2020-08-12:
[image]

ruspl-afed, 2020-08-12:
> Do you use 3rd party antivirus tools?

Windows defender, and it is not reporting any issue

ruspl-afed, 2020-08-12:
I killed all the python-related processes in the system and launched `pio home` again
Now it says 
`CorePackageManager: Installing contrib-pysite @ ~2.37.0`
it is much better, however I have no idea how long it can take

ivankravets, 2020-08-12:
It seems that I know the root of the issue now. Thanks! VSCode didn't kill PIO Home daemon at the exit which tried to install dependent packages. When PIO Core installs packages it LOCKs target directory. This is why it hangs. It's actually a bug and we'll think how to fix it. It could happen in a very rare case and I try to model a case that could reproduce this issue.

1. Kill all PYTHON.exe from system
2. Remove `.platformio`
3. Restart VSCode.

It should work now.

ivankravets, 2020-08-12:
Also, do you use a proxy?

ruspl-afed, 2020-08-12:
> Also, do you use a proxy?

No, I don't. However I can say that this week I have network problems from time to time. But I would expect the installer to fail by timeout and produce an error in this case.

ruspl-afed, 2020-08-12:
> It should work now.

I have the same situation. After killing and removing everything VSCode started its installation and it hangs as before. 
[image]