platformio/platformio-core-installer #56: UnicodeDecodeError: 'ascii' codec can't decode byte

GunZi200, 2020-08-24:
When I try to build my ATmega2560 project I get this on my Desktop PC. But it works fine on my laptop. Both using Windows 10 and same version of PlatformIO. Only thing I can think of is there are special characters **ð** and **á** in my windows user folder, but this has never been an issue before.

I am using PlatformIO 4.4.0a7 on both machines. VScode version 1.47.3

[image]

`pio upgrade --dev` and restarting VScode does not work.

platform.ini file:

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:micro]
platform = atmelavr
board = megaatmega2560
framework = arduino

; change microcontroller
board_build.mcu = atmega2560

; change MCU frequency
board_build.f_cpu = 16000000L

; baude rate for serial monitor
monitor_speed = 115200

; Libraries
lib_deps =
     https://github.com/arduino-libraries/Ethernet ; Ethernet library v2.0.0
     MQTT@2.4.7 ; MQTT library
     https://github.com/adafruit/RTClib ; Real-time clock library
     https://github.com/greiman/SdFat ; SD card library
     ; https://github.com/greiman/SdFat-beta

lib_ignore = TinyWireM
```

closed 2020-09-11 by ivankravets

ivankravets, 2020-09-11:
@GunZi200, @fengbaochun, @Ozakest, @Roman-ru

Please download https://raw.githubusercontent.com/platformio/platformio-core-installer/develop/get-platformio.py as save as `get-platformio.py`. Run this script with (please replace `REPLACE_WITH_YOUR_NAME`):

```
C:/Users/REPLACE_WITH_YOUR_NAME/.platformio/python3/python.exe path/to/downloaded/get-platformio.py
``` 

Does it work now?

GunZi200, 2020-09-12:
Can confirm the `get-platformio.py` script now works correctly on my end. Though I am having trouble running it on the Desktop PC. I get the following on my Laptop. Running` pio system info` shows Python 3 :)
[image]

On the Desktop PC I had some trouble with opening PlatformIO Home (In VScode) so I uninstalled the VScode plugin and removed the `.platformio` folder. Nothing happened until I opened an actual PlatformIO Project which is strange? Now when PlatformIO plugin tries to install the plugin I get a `UnicodeDecodeError`. I will post it in another comment below, not sure if that's a separate issue. 

@ivankravets Is it OK that on my Laptop `.platformio` folder is installed at `C:/Users:/(USER NAME)/.platformio` but on the Desktop PC its `C:/.platformio`? There the (USER NAME) is "GuðniMár" but on my Laptop its "Gudni". Just wondering if the 'ð' and 'á' characters are having any effect on the behaviour.

GunZi200, 2020-09-12:
@ivankravets This is the message I get on my Desktop PC in case it helps.

----------------------

https://github.com/platformio/platformio-core-installer/issues/new?title=Installation Manager&body=%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue+

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 69, in <module>
    main()
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 61, in main
    bootstrap()
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\.platformio\.cache\tmp\.piocore-installer-azij5pj4\tmp04oqbshs\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\.platformio\.cache\tmp\.piocore-installer-azij5pj4\tmp04oqbshs\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\.platformio\.cache\tmp\.piocore-installer-azij5pj4\tmp04oqbshs\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\.platformio\.cache\tmp\.piocore-installer-azij5pj4\tmp04oqbshs\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-azij5pj4\tmp04oqbshs\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-azij5pj4\tmp04oqbshs\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-azij5pj4\tmp04oqbshs\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\.platformio\.cache\tmp\.piocore-installer-azij5pj4\tmp04oqbshs\pioinstaller.zip\pioinstaller\__main__.py", line 52, in cli
  File "C:\.platformio\.cache\tmp\.piocore-installer-azij5pj4\tmp04oqbshs\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core
  File "C:\.platformio\.cache\tmp\.piocore-installer-azij5pj4\tmp04oqbshs\pioinstaller.zip\pioinstaller\penv.py", line 52, in create_core_penv
  File "C:\.platformio\.cache\tmp\.piocore-installer-azij5pj4\tmp04oqbshs\pioinstaller.zip\pioinstaller\python.py", line 174, in find_compatible_pythons
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf0 in position 29: invalid continuation byte
 -> Installer version: 0.3.4
Platform: Windows-10
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\.platformio\python3\python.exe
Creating a virtual environment at C:\.platformio\penv

	at c:\Users\GuðniMár\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4911
	at ChildProcess.c (c:\Users\GuðniMár\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\GuðniMár\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```
  &labels=auto

Roman-ru, 2020-09-13:
Yes, it works, thanks, but I had to download marlin 2.0.6.1 the other does
not build

ivankravets, 2020-09-13:
Thanks for the report! We will release new installer soon.

commit 3cb0154f (2020-09-11): Catch "UnicodeDecodeError" when outputting process result // Resolve #56