platformio/platformio-core-installer #57: Installation Manager

HTTP-0DAY, 2020-09-10:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.48.2
  PIO IDE: v2.0.0
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.4
Platform: Windows-10
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\Users\ma.dalaa\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\ma.dalaa\.platformio\penv

	at c:\Users\ma.dalaa\.vscode\extensions\platformio.platformio-ide-2.0.0\node_modules\platformio-node-helpers\dist\index.js:1:4911
	at ChildProcess.c (c:\Users\ma.dalaa\.vscode\extensions\platformio.platformio-ide-2.0.0\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:203:13)
	at ChildProcess.cp.emit (c:\Users\ma.dalaa\.vscode\extensions\platformio.platformio-ide-2.0.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-09-11:
Please run the next command in your system terminal and provide an output of:

```
C:/Usersma.dalaa/.platformio/python3/python.exe C:/Usersma.dalaa/.platformio/.cache/tmp/get-platformio-0.3.4.py
```

HTTP-0DAY, 2020-09-11:
Hello, i unistalled it and then i restarted my Computer and it woked ! thank you !

ivankravets, 2020-09-11:
Happy coding with PlatformIO!

closed 2020-09-11 by ivankravets