platformio/platformio-core-installer #61: Installation Manager: Platform update fails when `~/.platformio/` directory is a symlink

alaincao, 2020-09-10:
# Description of problem

The VSCode's installer failed to install/upgrade the environment on my computer (nb: installations/upgrades worked fine until today's upgrade).  
I could fix the issue by temporarily making my `~/.platformio/` a real directory instead of a symlink (ie. pointing to another mount point).  
When I revert back to the symlink situation after a successful installation, the environment works prefectly. So the problem is only with the installer.  

# Configuration

  VSCode: 1.48.2
  PIO IDE: v2.0.1
  System: Linux, 5.4.0-47-generic, x64
Ubuntu 20.04

# Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.4
Platform: Linux-5.4.0-47-generic-x86_64-with-glibc2.29
Python version: 3.8.2 (default, Jul 16 2020, 14:00:26) 
[GCC 9.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/aa/.platformio/penv

	at /mnt/poolnvme/works/_vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4911
	at ChildProcess.c (/mnt/poolnvme/works/_vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4807)
	at ChildProcess.emit (events.js:203:13)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

alaincao, 2020-09-10:
#### Out of topic note
I do remotely backup my home directory and so need to watch it's size closely.  
I am just really annoyed by each and every programs on earth putting their gigabytes of various files daily in it.  
My home directory is not a dumpster...

ivankravets, 2020-09-11:
Thanks for the report. We will investigate this issue. Meanwhile, you can move `.platformio` in another place via https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_CORE_DIR

ivankravets, 2020-09-16:
Sorry, I can't reproduce this issue. I created symlink 
```
ln -s /Users/ikravets/Downloads/.platformio/ /Users/ikravets/.platformio
```

```
PlatformIO Core has been successfully installed into an isolated environment `/Users/ikravets/.platformio/penv`!
```

alaincao, 2020-09-16:
Dear,

FYI, I cannot reproduce the issue anymore either.  
I even tried with my `~/.platformio` directory restored as it was that precise day, and somehow, the issue seem now resolved.  

So I suppose this issue can be definitively closed, now.
Thank you.

&nbsp;
&nbsp;

btw, sorry for my rant, that problem hit me at a particular bad moment ...

closed 2020-09-16 by alaincao

ivankravets, 2020-09-16:
Great! Happy coding with PlatformIO! 😊