platformio/platformio-core-installer #62: Installation Manager

victorfernandesraton, 2020-09-10:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.48.2
  PIO IDE: v2.0.1
  System: Linux, 5.4.0-45-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/victor/.platformio/.cache/tmp/get-platformio-0.3.4.py", line 69, in <module>
    main()
  File "/home/victor/.platformio/.cache/tmp/get-platformio-0.3.4.py", line 61, in main
    bootstrap()
  File "/home/victor/.platformio/.cache/tmp/get-platformio-0.3.4.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/home/victor/.platformio/.cache/tmp/.piocore-installer-bbrdi0ue/tmpzoxxivez/pioinstaller.zip/pioinstaller/__main__.py", line 122, in main
  File "/home/victor/.platformio/.cache/tmp/.piocore-installer-bbrdi0ue/tmpzoxxivez/pioinstaller.zip/click/core.py", line 829, in __call__
  File "/home/victor/.platformio/.cache/tmp/.piocore-installer-bbrdi0ue/tmpzoxxivez/pioinstaller.zip/click/core.py", line 782, in main
  File "/home/victor/.platformio/.cache/tmp/.piocore-installer-bbrdi0ue/tmpzoxxivez/pioinstaller.zip/click/core.py", line 1236, in invoke
  File "/home/victor/.platformio/.cache/tmp/.piocore-installer-bbrdi0ue/tmpzoxxivez/pioinstaller.zip/click/core.py", line 1066, in invoke
  File "/home/victor/.platformio/.cache/tmp/.piocore-installer-bbrdi0ue/tmpzoxxivez/pioinstaller.zip/click/core.py", line 610, in invoke
  File "/home/victor/.platformio/.cache/tmp/.piocore-installer-bbrdi0ue/tmpzoxxivez/pioinstaller.zip/click/decorators.py", line 21, in new_func
  File "/home/victor/.platformio/.cache/tmp/.piocore-installer-bbrdi0ue/tmpzoxxivez/pioinstaller.zip/pioinstaller/__main__.py", line 52, in cli
  File "/home/victor/.platformio/.cache/tmp/.piocore-installer-bbrdi0ue/tmpzoxxivez/pioinstaller.zip/pioinstaller/core.py", line 72, in install_platformio_core
  File "/home/victor/.platformio/.cache/tmp/.piocore-installer-bbrdi0ue/tmpzoxxivez/pioinstaller.zip/pioinstaller/penv.py", line 75, in create_core_penv
  File "/home/victor/.platformio/.cache/tmp/.piocore-installer-bbrdi0ue/tmpzoxxivez/pioinstaller.zip/pioinstaller/penv.py", line 144, in add_state_info
  File "/home/victor/anaconda3/lib/python3.7/subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "/home/victor/anaconda3/lib/python3.7/subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/home/victor/.platformio/penv/bin/python', '-c', "import sys

ivankravets, 2020-09-11:
Please run the next command in your system terminal and provide an output of:

```
python3 //home/victor/.platformio/.cache/tmp/get-platformio-0.3.4.py
```

victorfernandesraton, 2020-09-12:
> Please run the next command in your system terminal and provide an output of:
> 
> ```
> python3 //home/victor/.platformio/.cache/tmp/get-platformio-0.3.4.py
> ```

it's works, thanks

ivankravets, 2020-09-12:
The issue is caused by Anaconda software. We ignore it but it seems that we can't detect it on Linux. Please provide an output of these commands:
```
python3 --version
python3 -c "import sys;print(sys.executable, sys.version)"
```

closed 2020-10-16 by ivankravets