platformio/platformio-core-installer #65: Installation Manager

MapleHan, 2020-09-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\win\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 69, in <module>
    main()
  File "C:\Users\win\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 61, in main
    bootstrap()
  File "C:\Users\win\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\win\.platformio\.cache\tmp\.piocore-installer-67pxlw9w\tmpyd5a1sb0\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\Users\win\.platformio\.cache\tmp\.piocore-installer-67pxlw9w\tmpyd5a1sb0\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\Users\win\.platformio\.cache\tmp\.piocore-installer-67pxlw9w\tmpyd5a1sb0\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\Users\win\.platformio\.cache\tmp\.piocore-installer-67pxlw9w\tmpyd5a1sb0\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\Users\win\.platformio\.cache\tmp\.piocore-installer-67pxlw9w\tmpyd5a1sb0\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\Users\win\.platformio\.cache\tmp\.piocore-installer-67pxlw9w\tmpyd5a1sb0\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\Users\win\.platformio\.cache\tmp\.piocore-installer-67pxlw9w\tmpyd5a1sb0\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\Users\win\.platformio\.cache\tmp\.piocore-installer-67pxlw9w\tmpyd5a1sb0\pioinstaller.zip\pioinstaller\__main__.py", line 52, in cli
  File "C:\Users\win\.platformio\.cache\tmp\.piocore-installer-67pxlw9w\tmpyd5a1sb0\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core
  File "C:\Users\win\.platformio\.cache\tmp\.piocore-installer-67pxlw9w\tmpyd5a1sb0\pioinstaller.zip\pioinstaller\penv.py", line 75, in create_core_penv
  File "C:\Users\win\.platformio\.cache\tmp\.piocore-installer-67pxlw9w\tmpyd5a1sb0\pioinstaller.zip\pioinstaller\penv.py", line 144, in add_state_info
  File "C:\Users\win\.platformio\python3\lib\subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "C:\Users\win\.platformio\python3\lib\subprocess.py", line 488, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\win\.platformio\python3\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Users\win\.platformio\python3\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] ϵͳ�Ҳ���ָ�����ļ���
 -> Installer version: 0.3.4
Platform: Windows-10
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\Users\win\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\win\.platformio\penv

	at c:\Users\win\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4911
	at ChildProcess.c (c:\Users\win\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\win\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-09-11:
Please run the next command in your system terminal and provide an output of:

```
C:/Users/win/.platformio/python3/python.exe C:/Users/win/.platformio/.cache/tmp/get-platformio-0.3.4.py
```

Heptoxide, 2020-09-27:
[Manually running the get-platformio script worked for me](https://github.com/platformio/platformio-core-installer/issues/154#issuecomment-699666575)

f1oat, 2020-11-25:
Yes, problem solved with the command ".platformio\python3\python.exe .\.platformio\.cache\tmp\get-platformio-0.3.5.py"
As described in #154 
Thank you!

wrgallo, 2021-01-27:
> [Manually running the get-platformio script worked for me](https://github.com/platformio/platformio-core-installer/issues/154#issuecomment-699666575)

That did **not** worked for me, installing [GIT](https://git-scm.com/) and restarting VSCode worked.

closed 2021-03-06 by stale[bot]