platformio/platformio-core-installer #68: Installation Manager

KamurenRickC137, 2020-09-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.46.1
  PIO IDE: v2.0.1
  System: Windows_NT, 10.0.10240, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.4
Platform: Windows-10
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\Users\Jacob\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\Jacob\.platformio\penv

    at c:\Users\Jacob\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:138:23
    at ChildProcess.c (c:\Users\Jacob\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:124:5)
    at ChildProcess.emit (events.js:203:13)
    at ChildProcess.EventEmitter.emit (domain.js:476:20)
    at ChildProcess.cp.emit (c:\Users\Jacob\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1021:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

KamurenRickC137, 2020-09-11:
npm install issues

ivankravets, 2020-09-11:
Please run the next command in your system terminal and provide an output of:

```
C:/Users/Jacob/.platformio/python3/python.exe C:/Users/Jacob/.platformio/.cache/tmp/get-platformio-0.3.4.py
```

closed 2020-11-22 by stale[bot]