platformio/platformio-core-installer #69: Installation Manager

onlygille, 2020-09-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Darwin, 19.6.0, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.4
Platform: macOS-10.15.6
Python version: 3.8.4 (default, Jul 14 2020, 15:19:59) 
[Clang 10.0.0 (clang-1000.10.44.4)]
Python path: /Users/gilles/.platformio/python3/bin/python3
Creating a virtual environment at /Users/gilles/.platformio/penv

	at /Users/gilles/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4911
	at ChildProcess.c (/Users/gilles/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:223:5)
	at Pipe.<anonymous> (net.js:664:12)
  ```

ivankravets, 2020-09-11:
Please run the next command in your system terminal and provide an output of:

```
/Users/gilles/.platformio/python3/bin/python3 /Users/gilles/.platformio/.cache/tmp/get-platformio-0.3.4.py
```

onlygille, 2020-09-11:
Hey, thanks for the help. this is the output I got:

Installer version: 0.3.4
Platform: macOS-10.15.6
Python version: 3.8.4 (default, Jul 14 2020, 15:19:59)
[Clang 10.0.0 (clang-1000.10.44.4)]
Python path: /Users/gilles/.platformio/python3/bin/python3
Creating a virtual environment at /Users/gilles/.platformio/penv
Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues

ivankravets, 2020-09-11:
Could you repeat in verbose mode?
```
/Users/gilles/.platformio/python3/bin/python3 /Users/gilles/.platformio/.cache/tmp/get-platformio-0.3.4.py --verbose
```

onlygille, 2020-09-11:
Sure!

Installer version: 0.3.4
Platform: Darwin-19.6.0
Python version: 2.7.16 (default, Jun  5 2020, 22:59:21)
[GCC 4.2.1 Compatible Apple LLVM 11.0.3 (clang-1103.0.29.20) (-macos10.15-objc-
Python path: /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Creating a virtual environment at /Users/gilles/.platformio/penv
DEBUG: Current environment PATH /usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:/usr/local/go/bin:/Library/Apple/usr/bin:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Users/gilles/software/flutter/bin
DEBUG: Checking a Python candidate /usr/bin/python3
DEBUG: The Python 3.7.3 (/Applications/Xcode.app/Contents/Developer/usr/bin/python3) interpreter is compatible.
DEBUG: Checking a Python candidate /usr/bin/python
DEBUG: The Python 2.7.16 (/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python) interpreter is compatible.
DEBUG: Checking a Python candidate /usr/bin/python2
DEBUG: The Python 2.7.16 (/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python) interpreter is compatible.
DEBUG: Checking a Python candidate /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
DEBUG: The Python 2.7.16 (/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python) interpreter is compatible.
DEBUG: Using /usr/bin/python3 Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python3 -m venv /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv -p /usr/bin/python3 /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python3 /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: virtualenv /Users/gilles/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python3 /Users/gilles/.platformio/.cache/tmp/virtualenv.pyz /Users/gilles/.platformio/penv
DEBUG: Could not create virtualenv with downloaded script. Error: Command '['/usr/bin/python3', '/Users/gilles/.platformio/.cache/tmp/virtualenv.pyz', '/Users/gilles/.platformio/penv']' returned non-zero exit status 2
DEBUG: Using /usr/bin/python Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python -m venv /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python -m virtualenv -p /usr/bin/python /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python -m virtualenv /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: virtualenv /Users/gilles/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python /Users/gilles/.platformio/.cache/tmp/virtualenv.pyz /Users/gilles/.platformio/penv
DEBUG: Could not create virtualenv with downloaded script. Error: Command '['/usr/bin/python', '/Users/gilles/.platformio/.cache/tmp/virtualenv.pyz', '/Users/gilles/.platformio/penv']' returned non-zero exit status 2
DEBUG: Using /usr/bin/python2 Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python2 -m venv /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python2 -m virtualenv -p /usr/bin/python2 /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python2 /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python2 -m virtualenv /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: virtualenv /Users/gilles/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python2 /Users/gilles/.platformio/.cache/tmp/virtualenv.pyz /Users/gilles/.platformio/penv
DEBUG: Could not create virtualenv with downloaded script. Error: Command '['/usr/bin/python2', '/Users/gilles/.platformio/.cache/tmp/virtualenv.pyz', '/Users/gilles/.platformio/penv']' returned non-zero exit status 2
DEBUG: Using /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python Python for virtual environment.
DEBUG: Creating virtual environment: /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -m venv /Users/gilles/.platformio/penv
DEBUG: Creating virtual environment: /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -m virtualenv -p /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python /Users/gilles/.platformio/penv
[... cut ...]
DEBUG: Creating virtual environment: /Users/gilles/.platformio/python3/bin/python3 /Users/gilles/.platformio/.cache/tmp/virtualenv.pyz /Users/gilles/.platformio/penv
DEBUG: Could not create virtualenv with downloaded script. Error: Command '['/Users/gilles/.platformio/python3/bin/python3', '/Users/gilles/.platformio/.cache/tmp/virtualenv.pyz', '/Users/gilles/.platformio/penv']' returned non-zero exit status 2
Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues

ivankravets, 2020-09-11:
Something is going wrong with your machine. Could you remove `/Users/gilles/.platformio` folder and restart VSCode?

onlygille, 2020-09-11:
done

onlygille, 2020-09-11:
it gives me the same output :/

ivankravets, 2020-09-11:
Please contact me directly via me@ikravets.com and I'll provide further instructions on how to debug this issue.

onlygille, 2020-09-11:
ok

closed 2020-09-21 by onlygille