platformio/platformio-core-installer #70: Installation Manager

benstev, 2020-09-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Darwin, 16.7.0, x64

  %23 Exception
  ```
  Error: dyld: lazy symbol binding failed: Symbol not found: _utimensat
  Referenced from: /Users/bernard/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _utimensat
  Referenced from: /Users/bernard/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

Error: Could not install PlatformIO Core: Command '['/Users/bernard/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' died with <Signals.SIGABRT: 6>.
 -> Installer version: 0.3.4
Platform: macOS-10.12.6
Python version: 3.8.4 (default, Jul 14 2020, 15:19:59) 
[Clang 10.0.0 (clang-1000.10.44.4)]
Python path: /Users/bernard/.platformio/python3/bin/python3
Creating a virtual environment at /Users/bernard/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Using cached platformio-5.0.1.tar.gz (211 kB)

    at args (/Users/bernard/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:141:25)
    at ChildProcess.c (/Users/bernard/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:132:7)
    at ChildProcess.emit (events.js:223:5)
    at ChildProcess.EventEmitter.emit (domain.js:475:20)
    at maybeClose (internal/child_process.js:1021:16)
    at Socket.<anonymous> (internal/child_process.js:430:11)
    at Socket.emit (events.js:223:5)
    at Socket.EventEmitter.emit (domain.js:475:20)
    at Pipe.<anonymous> (net.js:664:12)
  ```

ivankravets, 2020-09-11:
Could you provide an output of this command?
```
/Users/bernard/.platformio/python3/bin/python3 -c "import os; print(os.stat('/Users'))"
```

benstev, 2020-09-12:
Sure, here it is:

`os.stat_result(st_mode=16877, st_ino=168857, st_dev=16777220, st_nlink=8, st_uid=0, st_gid=80, st_size=272, st_atime=1599813065, st_mtime=1577439242, st_ctime=1577439242)`

ivankravets, 2020-09-12:
Could you help to resolve this issue? We need to know which Python function fails on your machine. Then we can ignore pre-built Python.

You can also try:

```
/Users/bernard/.platformio/python3/bin/python3 /Users/bernard/.platformio/.cache/tmp/get-platformio-0.3.4.py --verbose
```

Please provide an output.

benstev, 2020-09-12:
here you go:
`
MacOS_Bernard:~ bernard$ /Users/bernard/.platformio/python3/bin/python3 /Users/bernard/.platformio/.cache/tmp/get-platformio-0.3.4.py --verbose
Installer version: 0.3.4
Platform: macOS-10.12.6
Python version: 3.8.4 (default, Jul 14 2020, 15:19:59) 
[Clang 10.0.0 (clang-1000.10.44.4)]
Python path: /Users/bernard/.platformio/python3/bin/python3
Creating a virtual environment at /Users/bernard/.platformio/penv
DEBUG: Current environment PATH /usr/local/sbin:/Users/bernard/Documents/google-cloud-sdk/bin:/Users/bernard/Documents/mongodb-osx-x86_64-3.4.10/bin:/usr/local/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Users/bernard/Documents/ffmpeg-20200115-0dc0837-macos64-static/bin:/opt/X11/bin:/Applications/Wireshark.app/Contents/MacOS
DEBUG: Checking a Python candidate /Users/bernard/.platformio/python3/bin/python3
DEBUG: The Python 3.8.4 (/Users/bernard/.platformio/python3/bin/python3) interpreter is compatible.
DEBUG: Checking a Python candidate /usr/local/bin/python3
DEBUG: The Python 3.7.7 (/usr/local/opt/python/bin/python3.7) interpreter is compatible.
DEBUG: Checking a Python candidate /usr/local/bin/python3
DEBUG: The Python 3.7.7 (/usr/local/opt/python/bin/python3.7) interpreter is compatible.
DEBUG: Checking a Python candidate /usr/bin/python
DEBUG: The Python 2.7.10 (/usr/bin/python) interpreter is compatible.
DEBUG: Using /Users/bernard/.platformio/python3/bin/python3 Python for virtual environment.
DEBUG: Creating virtual environment: /Users/bernard/.platformio/python3/bin/python3 -m venv /Users/bernard/.platformio/penv
DEBUG: Creating virtual environment: /Users/bernard/.platformio/python3/bin/python3 -m virtualenv -p /Users/bernard/.platformio/python3/bin/python3 /Users/bernard/.platformio/penv
DEBUG: Creating virtual environment: virtualenv -p /Users/bernard/.platformio/python3/bin/python3 /Users/bernard/.platformio/penv
DEBUG: Creating virtual environment: /Users/bernard/.platformio/python3/bin/python3 -m virtualenv /Users/bernard/.platformio/penv
DEBUG: Creating virtual environment: virtualenv /Users/bernard/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory: 'virtualenv'
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /Users/bernard/.platformio/python3/bin/python3 /Users/bernard/.platformio/.cache/tmp/virtualenv.pyz /Users/bernard/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
DEBUG: Creating pip.conf file in /Users/bernard/.platformio/penv
DEBUG: Downloading 'get-pip.py' installer...
DEBUG: Installing pip
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Using cached platformio-5.0.1.tar.gz (211 kB)
dyld: lazy symbol binding failed: Symbol not found: _utimensat
  Referenced from: /Users/bernard/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _utimensat
  Referenced from: /Users/bernard/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

Error: Could not install PlatformIO Core: Command '['/Users/bernard/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' died with <Signals.SIGABRT: 6>.
MacOS_Bernard:~ bernard$
`

ivankravets, 2020-09-13:
Please contact me@ikravets.com and will provide further details on how to debug this issue. Thanks

benstev, 2020-09-14:
Hello Ivan
It seems I finally made my way through by running the installer manually
with a local 3.8 python version.
I suppose the embedded python is not backward compatible with my MacOs
Sierra version.
Does this make any sense to you?
Thanks & regards
Bernard

closed 2020-09-14 by ivankravets

brianjpatterson, 2022-05-22:
So what happened with this?  Did you guys find a fix?   Sorry I don't understand github too well.   I got this same error installing shfmt on mac osx using brew install shfmt so its got nothing to do with python

commit e22a0354 (2020-09-14): Added workaround for "dyld: Symbol not found: _utimensat" issue for macOS < 10.13 // Resolve #70