platformio/platformio-core-installer #74: Installation Manager

HamzaHajeir, 2020-09-11:
### Description of problem:
PIO was running slightly until suddenly I got this message : 

### Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Windows_NT, 10.0.18362, x64

### Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\Hamza\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 69, in <module>
    main()
  File "C:\Users\Hamza\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 61, in main
    bootstrap()
  File "C:\Users\Hamza\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\Hamza\.platformio\.cache\tmp\.piocore-installer-xoelto3z\tmpr8rqxrcm\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\Users\Hamza\.platformio\.cache\tmp\.piocore-installer-xoelto3z\tmpr8rqxrcm\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\Users\Hamza\.platformio\.cache\tmp\.piocore-installer-xoelto3z\tmpr8rqxrcm\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\Users\Hamza\.platformio\.cache\tmp\.piocore-installer-xoelto3z\tmpr8rqxrcm\pioinstaller.zip\click\core.py", line 1259, in invoke
  File "C:\Users\Hamza\.platformio\.cache\tmp\.piocore-installer-xoelto3z\tmpr8rqxrcm\pioinstaller.zip\click\core.py", line 1259, in invoke
  File "C:\Users\Hamza\.platformio\.cache\tmp\.piocore-installer-xoelto3z\tmpr8rqxrcm\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\Users\Hamza\.platformio\.cache\tmp\.piocore-installer-xoelto3z\tmpr8rqxrcm\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\Users\Hamza\.platformio\.cache\tmp\.piocore-installer-xoelto3z\tmpr8rqxrcm\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\Users\Hamza\.platformio\.cache\tmp\.piocore-installer-xoelto3z\tmpr8rqxrcm\pioinstaller.zip\pioinstaller\__main__.py", line 106, in core_check
  File "C:\Users\Hamza\.platformio\.cache\tmp\.piocore-installer-xoelto3z\tmpr8rqxrcm\pioinstaller.zip\pioinstaller\core.py", line 193, in check
  File "C:\Users\Hamza\.platformio\python3\lib\subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "C:\Users\Hamza\.platformio\python3\lib\subprocess.py", line 488, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\Hamza\.platformio\python3\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Users\Hamza\.platformio\python3\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
PermissionError: [WinError 5] Access is denied

	at c:\Users\Hamza\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4911
	at ChildProcess.c (c:\Users\Hamza\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\Hamza\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

HamzaHajeir, 2020-09-11:
Update : PIO tries to re-install and hit an exception : 

> Installing PlatformIO IDE...
> It may take a few minutes depending on your connection speed
> Please do not close this window and do not open other folders until this process is completed.
> 
> Debugging information is available via VSCode > Help > Toggle Developer Tools > Console.
> 

And this comes after Restarting VSCode after this message when issuing a build command: 

The terminal process failed to launch: A native exception occurred during launch (Cannot create process, error code: 5).

HamzaHajeir, 2020-09-11:
I don't know whether it's the fault of anti-virus or not.. But disabling it made the build work again.

Note : PIO was running slightly with anti-virus enabled too. This seems new issue to me.

Thanks

ivankravets, 2023-05-31:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/884

closed 2023-05-31 by ivankravets