platformio/platformio-core-installer #75: Installation Manager

danilogf91, 2020-09-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Linux, 5.4.0-26-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "<frozen zipimport>", line 520, in _get_decompress_func
ModuleNotFoundError: No module named 'zlib'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen zipimport>", line 568, in _get_data
  File "<frozen zipimport>", line 523, in _get_decompress_func
zipimport.ZipImportError: can't decompress data

ivankravets, 2020-09-12:
What is your `python3 --version`?

closed 2020-11-22 by stale[bot]

danielpza, 2021-11-16:
> `python3 --version`

Hey I'm also having the same issue, my version is 

```
$ python3 --version
Python 3.10.0
```

ivankravets, 2021-11-16:
@danielpza what is your OS?

danielpza, 2021-11-16:
fedora 35, I fixed the issue shortly after by installing openssl devel libraries. So I guess it's not an issue with this project:

```sh
sudo dnf install openssl-devel
```