platformio/platformio-core-installer #82: Installation Manager

btmurrell, 2020-09-12:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Linux, 4.15.0-117-generic, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.4
Platform: Linux-4.15.0-117-generic-x86_64-with-glibc2.25
Python version: 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/merl/.platformio/penv

	at /home/merl/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4911
	at ChildProcess.c (/home/merl/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

thefat32, 2020-09-13:
Please provide an output from a system Terminal

```sh
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

ivankravets, 2020-09-14:
Please run the next command in your system terminal and provide an output of:

```
python3 ~.platformio/.cache/tmp/get-platformio-0.3.4.py
```

btmurrell, 2020-09-14:
thanks for your followup

first requested output:
```
$ echo $PATH
/home/merl/bin:/home/merl/n/bin:/home/merl/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/merl/bin/traveling-cf-admin:/usr/local/go/bin::/usr/local/go/bin:/home/merl/go/bin

$ echo $PYTHONPATH

$ python --version
Python 2.7.17

$ pip --version
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)

$ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
asn1crypto (0.24.0)
ccsm (0.9.13.1)
compizconfig-python (0.9.13.1)
cryptography (2.1.4)
enum34 (1.1.6)
idna (2.6)
ipaddress (1.0.17)
keyring (10.6.0)
keyrings.alt (3.0)
mutagen (1.42.0)
pdfshuffler (0.6.0)
pip (9.0.1)
powerline-status (2.6)
psutil (5.4.2)
pycairo (1.16.2)
pycrypto (2.6.1)
pygobject (3.26.1)
PyPDF2 (1.26.0)
pyxdg (0.25)
SecretStorage (2.3.1)
setuptools (39.0.1)
six (1.11.0)
wheel (0.30.0)
youtube-dl (2019.1.30.1)
youtube2mp3 (0.5.0)

```

second requested output:
```
$ python3 ~/.platformio/.cache/tmp/get-platformio-0.3.4.py
Installer version: 0.3.4
Platform: Linux-4.15.0-117-generic-x86_64-with-glibc2.25
Python version: 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/merl/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Processing ./.cache/pip/wheels/33/4b/95/a2174bd13c5ee256ac23ed04a8987e7824ae85449954042ea1/platformio-5.0.1-py3-none-any.whl
Collecting bottle<0.13
  Using cached bottle-0.12.18-py3-none-any.whl (89 kB)
Collecting click<8,>=5
  Using cached click-7.1.2-py2.py3-none-any.whl (82 kB)
Collecting pyserial!=3.3,<4,>=3
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting pyelftools<1,>=0.25
  Using cached pyelftools-0.26-py2.py3-none-any.whl (136 kB)
Collecting colorama
  Using cached colorama-0.4.3-py2.py3-none-any.whl (15 kB)
Collecting tabulate<1,>=0.8.3
  Using cached tabulate-0.8.7-py3-none-any.whl (24 kB)
Collecting marshmallow>=2
  Using cached marshmallow-3.7.1-py2.py3-none-any.whl (45 kB)
Collecting semantic-version<3,>=2.8.1
  Using cached semantic_version-2.8.5-py2.py3-none-any.whl (15 kB)
Collecting requests<3,>=2.4.0
  Using cached requests-2.24.0-py2.py3-none-any.whl (61 kB)
Collecting idna<3,>=2.5
  Using cached idna-2.10-py2.py3-none-any.whl (58 kB)
Collecting chardet<4,>=3.0.2
  Using cached chardet-3.0.4-py2.py3-none-any.whl (133 kB)
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1
  Using cached urllib3-1.25.10-py2.py3-none-any.whl (127 kB)
Collecting certifi>=2017.4.17
  Using cached certifi-2020.6.20-py2.py3-none-any.whl (156 kB)
Installing collected packages: bottle, click, pyserial, pyelftools, colorama, tabulate, marshmallow, semantic-version, idna, chardet, urllib3, certifi, requests, platformio
Successfully installed bottle-0.12.18 certifi-2020.6.20 chardet-3.0.4 click-7.1.2 colorama-0.4.3 idna-2.10 marshmallow-3.7.1 platformio-5.0.1 pyelftools-0.26 pyserial-3.4 requests-2.24.0 semantic-version-2.8.5 tabulate-0.8.7 urllib3-1.25.10

PlatformIO Core has been successfully installed into an isolated environment `/home/merl/.platformio/penv`!

The full path to `platformio.exe` is `/home/merl/.platformio/penv/bin/platformio`

If you need an access to `platformio.exe` from other applications, please install Shell Commands
(add PlatformIO Core binary directory `/home/merl/.platformio/penv/bin` to the system environment PATH variable):

See https://docs.platformio.org/page/installation.html#install-shell-commands

```

ivankravets, 2020-09-14:
Hm, it works now. Could you remove `/home/merl/.platformio/penv` folder and restart VSCode? I don't understand why it does not work in auto mode.

btmurrell, 2020-09-15:
I don't understand the state of things, but I did remove that `$HOME/.platformio/penv` folder and restarted vscode, and it went through an install sequence. I watched the C++ output window, and there were no errors. It prompted me to reload and I did. All seems good now. 

thanks

ivankravets, 2020-09-15:
Yes, a strange issue. Anyway, happy coding with PlatformIO!

closed 2020-09-15 by ivankravets