platformio/platformio-core-installer #85: Installation Manager

Yugo19, 2020-09-12:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Linux, 4.19.0-10-amd64, x64

  %23 Exception
  ```
  Error: Error: Can not install PlatformIO Core due to a missed `distutils` package in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-distutils

(MAY require administrator access `sudo`)
 -> Installer version: 0.3.4
Platform: Linux-4.19.0-10-amd64-x86_64-with-glibc2.28
Python version: 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/yugo/.platformio/penv

    at /home/yugo/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:138:23
    at ChildProcess.c (/home/yugo/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:124:5)
    at ChildProcess.emit (events.js:223:5)
    at ChildProcess.EventEmitter.emit (domain.js:475:20)
    at maybeClose (internal/child_process.js:1021:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

thefat32, 2020-09-13:
Have you tried `sudo apt-get install python3-distutils` ?

ivankravets, 2020-09-13:
Please run `sudo apt-get install python3-distutils` and restart VSCode.

DrDimedrol, 2020-12-16:
For me sudo apt-get install python3-venv help to install Platformio on linux. But I find the solution after 5 days of searching! I think it will be good idea if installer check presence of python3-venv package and display error. It will save a lot of time for a lot of people...

9ae8sdf76, 2021-02-22:
What's the command for MacOS Catalina? When I try to install PlatformIO on MacOS, I get the following errors:

```
Error: Error: Can not install PlatformIO Core due to a missed `distutils` package in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-distutils

(MAY require administrator access `sudo`)
 -> Installer version: 0.3.5
Platform: macOS-10.15.7
Python version: 3.9.1 (v3.9.1:1e5d33e9b9, Dec  7 2020, 12:10:52) 
[Clang 6.0 (clang-600.0.57)]
Python path: /Library/Frameworks/Python.framework/Versions/3.9/bin/python3
Creating a virtual environment at /Users/kstanl27/.platformio/penv

	at /Users/kstanl27/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275
	at Chi...
```

```
Error: Error: Can not install PlatformIO Core due to a missed `distutils` package in your Python installation. Please install this package manually using the OS package manager. For example:  $ apt-get install python3-distutils  (MAY require administrator access `sudo`)  -> Installer version: 0.3.5 Platform: macOS-10.15.7 Python version: 3.9.1 (v3.9.1:1e5d33e9b9, Dec  7 2020, 12:10:52)  [Clang 6.0 (clang-600.0.57)] Python path: /Library/Frameworks/Python.framework/Versions/3.9/bin/python3 Creating a virtual environment at /Users/kstanl27/.platformio/penv  	at /Users/kstanl27/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275 	at Chi...
onDidChangeNotification @ /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.desktop.main.js:738
/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.desktop.main.js:60 [Extension Host] Error: Error: Can not install PlatformIO Core due to a missed `distutils` package in your Python installation.Please install this package manually using the OS package manager. For example:$ apt-get install python3-distutils(MAY require administrator access `sudo`) -> Installer version: 0.3.5Platform: macOS-10.15.7Python version: 3.9.1 (v3.9.1:1e5d33e9b9, Dec  7 2020, 12:10:52) [Clang 6.0 (clang-600.0.57)]Python path: /Library/Frameworks/Python.framework/Versions/3.9/bin/python3Creating a virtual environment at /Users/kstanl27/.platformio/penv	at /Users/kstanl27/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275	at ChildProcess.c (/Users/kstanl27/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)	at ChildProcess.emit (events.js:315:20)	at maybeClose (internal/child_process.js:1021:16)	at Socket.<anonymous> (internal/child_process.js:443:11)	at Socket.emit (events.js:315:20)	at Pipe.<anonymous> (net.js:674:12)
b @ /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.desktop.main.js:60
```

Obviously there is no `sudo apt-get install` on mac. :)

SaeedHassan243, 2023-07-21:
Hi,
 I have even run the following commands to install req lib but still not working. It stucks when it says installing platformIO core and gives following error
 
 Error: Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)
 -> Installer version: 1.2.0
Platform: Linux-5.4.0-149-generic-x86_64-with-glibc2.27
Python version: 3.9.7 (default, Jul 21 2023, 15:26:20) 
[GCC 7.5.0]
Python path: /usr/local/bin/python3
Creating a virtual environment at /home/saeed/.platformio/penv

	at /home/saeed/.vscode/extensions/platformio.platformio-ide-3.3.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5300
	at ChildProcess.i (/home/saeed/.vscode/extensions/p...
	
	Just to be sure I have run all the commands mentioned above
	And I have also tried in two different ubuntu version
	Thanks,
	saeed

ivankravets, 2023-07-21:
@SaeedHassan243 could you check this thread https://community.platformio.org/t/platformio-ide-doesnt-see-python3-vevn-on-linux/34857/3 ?

James-Livesey, 2023-07-23:
For people who have Python versions above 3.8 installed, you might want to run this:

```bash
sudo apt remove python3.10 python3.10-dev python3.10-minimal python3.9-minimal
```

Worked for me, was suggested by @freynder: https://github.com/platformio/platformio-core-installer/issues/2149#issuecomment-1643505731

ivankravets, 2023-07-24:
You don't need to remove the newest Python. Jut install `apt install python3.10-venv`.

ivankravets, 2023-07-24:
See https://community.platformio.org/t/platformio-ide-doesnt-see-python3-vevn-on-linux/34857/10?u=ivankravets

freynder, 2023-07-24:
> For people who have Python versions above 3.8 installed, you might want to run this:
> 
> ```shell
> sudo apt remove python3.10 python3.10-dev python3.10-minimal python3.9-minimal
> ```
> 
> Worked for me, was suggested by @freynder: [#2149 (comment)](https://github.com/platformio/platformio-core-installer/issues/2149#issuecomment-1643505731)

That was just a quick solution, see @ivankravets comment below for a cleaner solution. Ideally the installer would be more clear though.

ivankravets, 2023-07-24:
Thanks, @freynder! I've just updated original comment https://github.com/platformio/platformio-core-installer/issues/85#issue-700409072

AeroSynth, 2023-07-24:
I'm trying to install platformio in Windows 10.  I unchecked "Proxy Strict SSL" as mentioned earlier.
But I get this error:

`Error: Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation. Please install this package manually using the OS package manager. For example: $ apt-get install python3-venv (MAY require administrator access `sudo`) -> Installer version: 1.2.0 Platform: Windows-10 Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)] Python path: C:\Users\brads\.platformio\python3\python.exe Creating a virtual environment at C:\Users\brads\.platformio\penv at c:\Users\brads\.vscode\extensions\platformio.platformio-ide-3.3.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5300 at ChildProcess.i...`

The "suggestion" show you how to do this in Linux but not in Windows.
Do I really need to run Python-venv?

ivankravets, 2023-07-25:
@AeroSynth do you use any espionage ("antivirus") software? It can block PlatformIO. Please try to remove `C:\Users\%USER_PROFILE%\.platformio` folder and restart VSCode.

patvdleer, 2023-07-27:
```
Error: Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)
 -> Installer version: 1.2.0
Platform: Linux-5.19.0-45-generic-x86_64-with-glibc2.35
Python version: 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/patrick/.platformio/penv

    at /home/patrick/.vscode/extensions/platformio.platformio-ide-3.3.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5300
    at ChildProcess.i (/home/patrick/.vscode/extensio...
```

`aptitude search python3 | grep venv | grep "^i"`
```
i  python3.8-venv - Interactive high-level object-oriented language (pyvenv binary, version 3.8)
i  python3.9-venv - Interactive high-level object-oriented language (pyvenv binary, version 3.9)
i  python3.10-venv - Interactive high-level object-oriented language (pyvenv binary, version 3.10)
i  python3.11-venv - Interactive high-level object-oriented language (pyvenv binary, version 3.11)
```

Manual check:
```
patrick@patrick-Yoga-Slim-7-Pro-14ACH5:~$ python3 -m venv -h
usage: venv [-h] [--system-site-packages] [--symlinks | --copies] [--clear] [--upgrade] [--without-pip] [--prompt PROMPT] [--upgrade-deps] ENV_DIR [ENV_DIR ...]
...
```

what am I missing?

ivankravets, 2023-07-27:
@patvdleer, does this command work for you?
```
rm -rf ~/.platformio/penv
python3 -m venv ~/.platformio/penv
```

ivankravets, 2023-07-27:
> Host system:

The installer works on the remote side (inside the container). Could you log in to the container shell and repeat the same commands? `python3 -m venv --help`, etc...

lukew151, 2023-07-27:
> > Host system:
> 
> The installer works on the remote side (inside the container). Could you log in to the container shell and repeat the same commands? `python3 -m venv --help`, etc...

I have further down in my original comment under `VS Code term (inside container)`

ivankravets, 2023-07-27:
Try the next inside container:
```
cd /tmp
python3 -m venv pioenv
```

Please share output here.

lukew151, 2023-07-27:
in the container
```
root@47e608987c3c:/workspaces/Marlin# cd /tmp
root@47e608987c3c:/tmp# python3 -m venv pioenv
root@47e608987c3c:/tmp# echo $?
0
```

ivankravets, 2023-07-27:
Now, please try PIO Core installer directly https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#super-quick-macos-linux

lukew151, 2023-07-27:
```
root@47e608987c3c:/tmp# python3 -m venv pioenv
root@47e608987c3c:/tmp# echo $?
0
root@47e608987c3c:/tmp# python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
Installer version: 1.2.0
Platform: Linux-6.1.0-10-amd64-x86_64-with-glibc2.36
Python version: 3.11.4 (main, Jul  4 2023, 05:25:16) [GCC 12.2.0]
Python path: /usr/local/bin/python3
Creating a virtual environment at /root/.platformio/penv
Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation.
```

ivankravets, 2023-07-27:
Hm... Please run the following:
```bash
cd /tmp
wget https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py
python3 get-platformio.py check
```

ivankravets, 2023-07-27:
Also, could you share instructions on how I can reproduce this issue on the Linux machine?

lukew151, 2023-07-27:
It's a desktop install of Debian Bookworm,  I don't think it has anything particuarly special installed.
```
root@47e608987c3c:/tmp# python3 get-platformio.py check core
Error: Compatible PlatformIO Core not found.
Reason: PlatformIO executable not found in `/root/.platformio/penv/bin`
Traceback (most recent call last):
  File "/tmp/get-platformio.py", line 105, in <module>
    sys.exit(main())
             ^^^^^^
  File "/tmp/get-platformio.py", line 101, in main
    subprocess.check_call(command)
  File "/usr/local/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/local/bin/python3', '/tmp/tmpea2fcvoh', 'check', 'core']' returned non-zero exit status 1.
root@47e608987c3c:/tmp# python3 get-platformio.py check python
The Python 3.11.4 (/usr/local/bin/python3) interpreter is compatible.
```

ivankravets, 2023-07-27:
Sorry, I fixed my comment. The command should be `python3 get-platformio.py check python`

lukew151, 2023-07-27:
OK, inside the container:
```
# apt update
# apt install python3-venv
# python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
```
So I had python-nenv installed in the host, but the container needed it installed as well.

ivankravets, 2023-07-27:
> So I had python-nenv installed in the host, but the container needed it installed as well.

I thought you installed it. Of course, it will not work without `python3.11-venv` in your case.

So, does it work now?

patvdleer, 2023-07-27:
> @patvdleer, does this command work for you?
> 
> ```
> rm -rf ~/.platformio/penv
> python3 -m venv ~/.platformio/penv
> ```

Same error...

```
patrick@patrick-Yoga-Slim-7-Pro-14ACH5:~$ rm -rf ~/.platformio/penv
patrick@patrick-Yoga-Slim-7-Pro-14ACH5:~$ python3 -m venv ~/.platformio/penv

### CHECKED VSCODE HERE ###

patrick@patrick-Yoga-Slim-7-Pro-14ACH5:~$ cd ~/.platformio/penv/
patrick@patrick-Yoga-Slim-7-Pro-14ACH5:~/.platformio/penv$ source bin/activate
(penv) patrick@patrick-Yoga-Slim-7-Pro-14ACH5:~/.platformio/penv$ wget https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py
--2023-07-27 23:20:43--  https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8000::154, 2606:50c0:8002::154, 2606:50c0:8003::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8000::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2731 (2,7K) [text/plain]
Saving to: ‘get-platformio.py’

get-platformio.py           100%[=========================================>]   2,67K  --.-KB/s    in 0s      

2023-07-27 23:20:43 (64,1 MB/s) - ‘get-platformio.py’ saved [2731/2731]

(penv) patrick@patrick-Yoga-Slim-7-Pro-14ACH5:~/.platformio/penv$ python3 get-platformio.py check python
The Python 3.10.6 (/home/patrick/.platformio/penv/bin/python3) interpreter is compatible.
(penv) patrick@patrick-Yoga-Slim-7-Pro-14ACH5:~/.platformio/penv$ echo $?
0
```

ivankravets, 2023-07-28:
@patvdleer , the `get-platformio.py` installer MUST BE RUN outside the virtual environment. So, this `source bin/activate` is a REDUNDANT step. Try 

```
rm -rf ~/.platformio/penv
python3 get-platformio.py
```

Mathe13, 2023-07-28:
> @patvdleer , the `get-platformio.py` installer MUST BE RUN outside the virtual environment. So, this `source bin/activate` is a REDUNDANT step. Try
> 
> ```
> rm -rf ~/.platformio/penv
> python3 get-platformio.py
> ```

when i run python3 get-platformio.py i get this error message 
`
Installer version: 1.2.0
Platform: Linux-5.10.0-1014-oem-x86_64-with-glibc2.29
Python version: 3.8.10 (default, May 26 2023, 14:05:08) 
[GCC 9.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/matheus/.platformio/penv
Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)

`
but when i run python3 get-platformio.py check python it says its all fine

`
The Python 3.8.10 (/usr/bin/python3) interpreter is compatible.

`
also i alreay installed python3-venv

ivankravets, 2023-07-28:
> The Python 3.8.10 

Have you read the original comment https://github.com/platformio/platformio-core-installer/issues/85#issue-700409072 ?

It must be the following command in your case with default Python 3.8

```
apt-get install python3.8-venv
```

Mathe13, 2023-07-28:
> > The Python 3.8.10
> 
> Have you read the original comment [#85 (comment)](https://github.com/platformio/platformio-core-installer/issues/85#issue-700409072) ?
> 
> It must be the following command in your case with default Python 3.8
> 
> ```
> apt-get install python3.8-venv
> ```

already used that comand

`pt search python3.8 | grep -e venv -e ensurepip

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

python3.8-venv/focal-updates,focal-security,now 3.8.10-0ubuntu1~20.04.8 amd64 [instalado]
  Interactive high-level object-oriented language (pyvenv binary, version 3.8)`

ivankravets, 2023-07-28:
Please run the following commands and provide output:
```
rm -rf ~/.platformio/penv
python3  get-platformio.py --verbose
```

Mathe13, 2023-07-28:
> Please run the following commands and provide output:
> 
> ```
> rm -rf ~/.platformio/penv
> python3  get-platformio.py --verbose
> ```

```
Installer version: 1.2.0
Platform: Linux-5.10.0-1014-oem-x86_64-with-glibc2.29
Python version: 3.8.10 (default, May 26 2023, 14:05:08) 
[GCC 9.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/matheus/.platformio/penv
DEBUG: Current environment PATH /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
DEBUG: Checking a Python candidate /usr/bin/python3.9
DEBUG: Error: The Python 3.9.5 (/usr/bin/python3.9) interpreter is not compatible.
Reason: Could not find Python `venv` module

Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)
Traceback (most recent call last):
  File "get-platformio.py", line 105, in <module>
    sys.exit(main())
  File "get-platformio.py", line 101, in main
    subprocess.check_call(command)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '/tmp/tmpdzdpyx9p', '--verbose']' returned non-zero exit status 1.
```

Mathe13, 2023-07-28:
removing python3.9 it worked

ivankravets, 2023-07-28:
We have improved the installer and now it ignores broken Python installations and uses the compatible Pythons. See https://github.com/platformio/platformio-core-installer/releases/tag/v1.2.1

Hope this release will fix the issue on Linux.

ivankravets, 2023-07-28:
PlatformIO IDE for VSCode 3.3.1 is out! https://github.com/platformio/platformio-vscode-ide/releases/tag/v3.3.1

Bernycomputer10, 2025-02-25:
sudo apt-get install python3-venv