platformio/platformio-core-installer #86: Installation Manager

el-quique, 2020-09-12:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Linux, 5.4.0-47-generic, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.4
Platform: Linux-5.4.0-47-generic-x86_64-with-glibc2.29
Python version: 3.8.2 (default, Jul 16 2020, 14:00:26) 
[GCC 9.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/el-quique/.platformio/penv

	at /extend/el-quique/ConfiguracionHome/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4911
	at ChildProcess.c (/extend/el-quique/ConfiguracionHome/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

thefat32, 2020-09-13:
Please provide an output from a system Terminal

```sh
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

ivankravets, 2020-09-14:
Please run the next command in your system terminal and provide an output of:

```
python3 ~.platformio/.cache/tmp/get-platformio-0.3.4.py
```

rtlopez, 2020-09-16:
Hi, I have the same issue

# Configuration

VSCode: 1.49.0 (latest)
PIO IDE: v2.1.0
System: Linux, 5.4.0-47-generic, x64 (Kubuntu 20.04)

# Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.5
Platform: Linux-5.4.0-47-generic-x86_64-with-glibc2.29
Python version: 3.8.2 (default, Jul 16 2020, 14:00:26) 
[GCC 9.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/romek/.platformio/penv

	at /srv/devel/projects/.vscode/extensions/platformio.platformio-ide-2.1.0/node_modules/platformio-node-helpers/dist/index.js:1:4911
	at ChildProcess.s (/srv/devel/projects/.vscode/extensions/platformio.platformio-ide-2.1.0/node_modules/platformio-node-helpers/dist/index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.EventEmitter.emit (domain.js:475:20)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```
if I run installer, I'm getting
```
python3 ~/.platformio/.cache/tmp/get-platformio-0.3.5.py
Installer version: 0.3.5
Platform: Linux-5.4.0-47-generic-x86_64-with-glibc2.29
Python version: 3.8.2 (default, Jul 16 2020, 14:00:26) 
[GCC 9.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/romek/.platformio/penv
Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
```
and then 
```
echo $PATH
/home/romek/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/lib/jvm/java-8-oracle/bin:/usr/lib/jvm/java-8-oracle/db/bin:/usr/lib/jvm/java-8-oracle/jre/bin
echo $PYTHONPATH

python --version
Python 2.7.18rc1
python3 --version
Python 3.8.2
pip --version
Command 'pip' not found, but there are 18 similar ones.
pip list
Command 'pip' not found, but there are 18 similar ones.
```
it seems thate there is no `pip` command, should I have it installed? or is it result of broken python environment?

Cheers.

ivankravets, 2020-09-17:
Could you try 

```
python3 ~/.platformio/.cache/tmp/get-platformio-0.3.5.py --verbose
```

What is your distributive? Did you install custom Python?

rtlopez, 2020-09-17:
Hi, here is a result
```
python3 ~/.platformio/.cache/tmp/get-platformio-0.3.5.py --verbose
Installer version: 0.3.5
Platform: Linux-5.4.0-47-generic-x86_64-with-glibc2.29
Python version: 3.8.2 (default, Jul 16 2020, 14:00:26) 
[GCC 9.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/romek/.platformio/penv
DEBUG: Current environment PATH /home/romek/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/lib/jvm/java-8-oracle/bin:/usr/lib/jvm/java-8-oracle/db/bin:/usr/lib/jvm/java-8-oracle/jre/bin
DEBUG: Checking a Python candidate /usr/bin/python3
DEBUG: The Python 3.8.2 (/usr/bin/python3) interpreter is compatible.
DEBUG: Checking a Python candidate /usr/bin/python
DEBUG: The Python 2.7.18rc1 (/usr/bin/python) interpreter is compatible.
DEBUG: Checking a Python candidate /usr/bin/python2
DEBUG: The Python 2.7.18rc1 (/usr/bin/python2) interpreter is compatible.
DEBUG: Using /usr/bin/python3 Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python3 -m venv /home/romek/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv -p /usr/bin/python3 /home/romek/.platformio/penv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python3 /home/romek/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python3 -m virtualenv /home/romek/.platformio/penv
DEBUG: Creating virtual environment: virtualenv /home/romek/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory: 'virtualenv'
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python3 /home/romek/.platformio/.cache/tmp/virtualenv.pyz /home/romek/.platformio/penv
DEBUG: Could not create virtualenv with downloaded script. Error: Command '['/usr/bin/python3', '/home/romek/.platformio/.cache/tmp/virtualenv.pyz', '/home/romek/.platformio/penv']' returned non-zero exit status 1.
DEBUG: Using /usr/bin/python Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python -m venv /home/romek/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python -m virtualenv -p /usr/bin/python /home/romek/.platformio/penv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python /home/romek/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python -m virtualenv /home/romek/.platformio/penv
DEBUG: Creating virtual environment: virtualenv /home/romek/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory: 'virtualenv'
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python /home/romek/.platformio/.cache/tmp/virtualenv.pyz /home/romek/.platformio/penv
DEBUG: Could not create virtualenv with downloaded script. Error: Command '['/usr/bin/python', '/home/romek/.platformio/.cache/tmp/virtualenv.pyz', '/home/romek/.platformio/penv']' returned non-zero exit status 1.
DEBUG: Using /usr/bin/python2 Python for virtual environment.
DEBUG: Creating virtual environment: /usr/bin/python2 -m venv /home/romek/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python2 -m virtualenv -p /usr/bin/python2 /home/romek/.platformio/penv
DEBUG: Creating virtual environment: virtualenv -p /usr/bin/python2 /home/romek/.platformio/penv
DEBUG: Creating virtual environment: /usr/bin/python2 -m virtualenv /home/romek/.platformio/penv
DEBUG: Creating virtual environment: virtualenv /home/romek/.platformio/penv
DEBUG: Could not create virtualenv with local packages Trying download virtualenv script and using it. Error: [Errno 2] No such file or directory: 'virtualenv'
DEBUG: Downloading virtualenv package archive
DEBUG: Creating virtual environment: /usr/bin/python2 /home/romek/.platformio/.cache/tmp/virtualenv.pyz /home/romek/.platformio/penv
DEBUG: Could not create virtualenv with downloaded script. Error: Command '['/usr/bin/python2', '/home/romek/.platformio/.cache/tmp/virtualenv.pyz', '/home/romek/.platformio/penv']' returned non-zero exit status 1.
Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
```

I'm using Kubuntu 20.04 (upgraded from 18.04). I've never installed any cutom python package on my system.
Cheers.

rtlopez, 2020-09-20:
I'm not sure if it makes sense, I've tried to run one command from debug above.

```
$ /usr/bin/python3 /home/romek/.platformio/.cache/tmp/virtualenv.pyz /home/romek/.platformio/penv
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/__main__.py", line 168, in <module>
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/__main__.py", line 164, in run
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/__main__.py", line 16, in run
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/run/__init__.py", line 28, in cli_run
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/run/session.py", line 46, in run
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/run/session.py", line 53, in _create
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/create/creator.py", line 157, in run
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/create/via_global_ref/builtin/via_global_self_do.py", line 86, in create
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/create/via_global_ref/api.py", line 88, in create
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/create/via_global_ref/api.py", line 91, in install_patch
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/create/via_global_ref/builtin/cpython/cpython3.py", line 27, in env_patch_text
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/create/via_global_ref/api.py", line 102, in env_patch_text
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/app_data/base.py", line 54, in ensure_extracted
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/app_data/via_disk_folder.py", line 81, in extract
  File "/home/romek/.platformio/.cache/tmp/virtualenv.pyz/virtualenv/util/zipapp.py", line 22, in extract
    
  File "/usr/lib/python3.8/zipfile.py", line 1441, in getinfo
    raise KeyError(
KeyError: "There is no item named 'nv/create/via_global_ref/_virtualenv.py' in the archive"
```
Cheers

xuanduc611, 2020-09-25:
> python3 ~/.platformio/.cache/tmp/get-platformio-0.3.5.py --verbose

This command-line works for me. 

My environment:
- Ubuntu 14.04.6
- Linux 4.4.0-148-generic x86_64
- python 2.7.9

rtlopez, 2020-10-04:
Hi,

I dug into this a bit, and it seems that when upgrading ubuntu from 18.04 to 20.04 python 2.7 is kept in system (new installations doesn't have it) https://wiki.ubuntu.com/FocalFossa/ReleaseNotes#Python3_by_default. So I purged all python2 packages from my system. Unfortunately it didn't help. As next shot I decided to install python3-virtualenv package `apt install python3-virtualenv`. After this step platformio instalation has been completed.

Cheers,

closed 2020-12-10 by stale[bot]