platformio/platformio-core-installer #90: Installation Manager

cristianro99, 2020-09-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Windows_NT, 10.0.18362, x64

  %23 Exception
  ```
  TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received type undefined
	at validateString (internal/validators.js:112:11)
	at Object.dirname (path.js:583:5)
	at t.value (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:29919)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async t.value (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:27850)
	at async t.default.install (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.0.1\dist\extension.js:1:22972)
	at async c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.0.1\dist\extension.js:1:6210
  ```

ivankravets, 2020-09-14:
Please run the next command in your system terminal and provide an output of:

```
C:/Users/Administrator/.platformio/python3/python.exe C:/Users/Administrator/.platformio/.cache/tmp/get-platformio-0.3.4.py
```

cristianro99, 2020-09-16:
hi, I run run this command :C:/Users/Administrator/.platformio/python3/python.exe and is ok, 
C:/Users/Administrator/.platformio/.cache/tmp/        get-platformio-0.3.4.py    does not exist.
Can you tell me how to install the interpreter?

ivankravets, 2020-09-17:
I updated the command above, the installer was updated to 0.3.5. Do you see it now?

closed 2020-11-23 by stale[bot]