platformio/platformio-core-installer #92: Installation Manager

M4ngu, 2020-09-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Darwin, 16.7.0, x64

  %23 Exception
  ```
  Error: dyld: lazy symbol binding failed: Symbol not found: _utimensat
  Referenced from: /Users/jesus/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _utimensat
  Referenced from: /Users/jesus/.platformio/penv/bin/python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib

Error: Could not install PlatformIO Core: Command '['/Users/jesus/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' died with <Signals.SIGABRT: 6>.
 -> Installer version: 0.3.4
Platform: macOS-10.12.6
Python version: 3.8.4 (default, Jul 14 2020, 15:19:59) 
[Clang 10.0.0 (clang-1000.10.44.4)]
Python path: /Users/jesus/.platformio/python3/bin/python3
Creating a virtual environment at /Users/jesus/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Using cached platformio-5.0.1.tar.gz (211 kB)

	at /Users/jesus/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4911
	at ChildProcess.c (/Users/jesus/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:223:5)
	at Pipe.<anonymous> (net.js:664:12)
  ```

ivankravets, 2020-09-14:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/70

closed 2020-09-14 by ivankravets

ivankravets, 2020-09-14:
Could you provide a remote session to debug this issue? Please ping me at me@ikravets.com.

M4ngu, 2020-09-14:
I would like to, not sure how to do that 'remote session' thing

ivankravets, 2020-09-14:
Please email at me@ikravets.com and I'll provide further details.