platformio/platformio-core-installer #94: Installation Manager

albronda, 2020-09-14:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.4
Platform: Windows-10
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\Users\albro\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\albro\.platformio\penv

    at c:\Users\albro\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:138:23
    at ChildProcess.c (c:\Users\albro\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:124:5)
    at ChildProcess.emit (events.js:223:5)
    at ChildProcess.cp.emit (c:\Users\albro\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1021:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-09-14:
Do you have antivirus tool in the system?

albronda, 2020-09-14:
Hi,

no I only use the default Defender form Windows 10.

Thank you for your reply,

Peter

closed 2020-11-22 by stale[bot]