platformio/platformio-core-installer #100: Installation Manager

alex95, 2020-09-15:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 69, in <module>
    main()
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 61, in main
    bootstrap()
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\.platformio\.cache\tmp\.piocore-installer-rj7i5eiz\tmpl4u2uae0\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\.platformio\.cache\tmp\.piocore-installer-rj7i5eiz\tmpl4u2uae0\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\.platformio\.cache\tmp\.piocore-installer-rj7i5eiz\tmpl4u2uae0\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\.platformio\.cache\tmp\.piocore-installer-rj7i5eiz\tmpl4u2uae0\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-rj7i5eiz\tmpl4u2uae0\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-rj7i5eiz\tmpl4u2uae0\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-rj7i5eiz\tmpl4u2uae0\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\.platformio\.cache\tmp\.piocore-installer-rj7i5eiz\tmpl4u2uae0\pioinstaller.zip\pioinstaller\__main__.py", line 52, in cli
  File "C:\.platformio\.cache\tmp\.piocore-installer-rj7i5eiz\tmpl4u2uae0\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core
  File "C:\.platformio\.cache\tmp\.piocore-installer-rj7i5eiz\tmpl4u2uae0\pioinstaller.zip\pioinstaller\penv.py", line 52, in create_core_penv
  File "C:\.platformio\.cache\tmp\.piocore-installer-rj7i5eiz\tmpl4u2uae0\pioinstaller.zip\pioinstaller\python.py", line 174, in find_compatible_pythons
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc0 in position 27: invalid start byte
 -> Installer version: 0.3.4
Platform: Windows-10
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\.platformio\python3\python.exe
Creating a virtual environment at C:\.platformio\penv

	at c:\Users\Администратор\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4911
	at ChildProcess.c (c:\Users\Администратор\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\Администратор\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-09-16:
Fixed in https://github.com/platformio/platformio-core-installer/issues/56

The final release on PIO IDE for VSCode is coming today.

closed 2020-09-16 by ivankravets

alex95, 2020-09-16:
I found the reason, all because of the username (it is in Russian)

ivankravets, 2020-09-16:
Yes, we fixed this. Give us 20-30mins and will release a new version