platformio/platformio-core-installer #101: platformio installation

MykhailKo, 2020-09-15:
Platformio stopped properly working after Windows update, tried to delete the old .platformio folder, and reinstalling VsCode, did not seem to help.
Running command 
C:/Users/Михаил/.platformio/python3/._python.exe C:/Users/Михаил/.platformio/.cache/tmp/get-platformio-0.3.4.py 
resulted in  
+ CategoryInfo          : ObjectNotFound: (C:/Users/Михаил...n3/._python.exe:String) [],
 CommandNotFoundException     + FullyQualifiedErrorId : CommandNotFoundException

%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 69, in <module>
    main()
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 61, in main
    bootstrap()
  File "C:\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\.platformio\.cache\tmp\.piocore-installer-m02swfvn\tmp9186ufgd\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\.platformio\.cache\tmp\.piocore-installer-m02swfvn\tmp9186ufgd\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\.platformio\.cache\tmp\.piocore-installer-m02swfvn\tmp9186ufgd\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\.platformio\.cache\tmp\.piocore-installer-m02swfvn\tmp9186ufgd\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-m02swfvn\tmp9186ufgd\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-m02swfvn\tmp9186ufgd\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-m02swfvn\tmp9186ufgd\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\.platformio\.cache\tmp\.piocore-installer-m02swfvn\tmp9186ufgd\pioinstaller.zip\pioinstaller\__main__.py", line 52, in cli
  File "C:\.platformio\.cache\tmp\.piocore-installer-m02swfvn\tmp9186ufgd\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core
  File "C:\.platformio\.cache\tmp\.piocore-installer-m02swfvn\tmp9186ufgd\pioinstaller.zip\pioinstaller\penv.py", line 52, in create_core_penv
  File "C:\.platformio\.cache\tmp\.piocore-installer-m02swfvn\tmp9186ufgd\pioinstaller.zip\pioinstaller\python.py", line 174, in find_compatible_pythons
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xcc in position 27: invalid continuation byte
 -> Installer version: 0.3.4
Platform: Windows-10
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\.platformio\python3\python.exe
Creating a virtual environment at C:\.platformio\penv

	at c:\Users\Михаил\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4911
	at ChildProcess.c (c:\Users\Михаил\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\Михаил\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-09-16:
Fixed in https://github.com/platformio/platformio-core-installer/issues/56

The final release on PIO IDE for VSCode is coming today.

closed 2020-09-16 by ivankravets

MykhailKo, 2020-09-16:
Thnak you!

ivankravets, 2020-09-16:
Будь ласка, обновіться на PlatformIO IDE 2.1.0 для VSCode. Деталі https://github.com/platformio/platformio-vscode-ide/releases/tag/v2.1.0