platformio/platformio-core-installer #105: Installation Manager - error during update

panciunio, 2020-09-16:
%23 Description of problem
Failed to install PlatformIO IDE. The error annouced that the OS can't install python3-distutils but unfortunatelly this package is installed properly: python3-distutils is already the newest version (3.8.2-1ubuntu1).

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Linux, 5.4.0-47-generic, x64

  %23 Exception
  ```
  Error: Error: Can not install PlatformIO Core due to a missed `distutils` package in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-distutils

(MAY require administrator access `sudo`)
 -> Installer version: 0.3.4
Platform: Linux-5.4.0-47-generic-x86_64-with-glibc2.29
Python version: 3.8.2 (default, Jul 16 2020, 14:00:26) 
[GCC 9.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/pancio/.platformio/penv

	at /home/pancio/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4911
	at ChildProcess.c (/home/pancio/.vscode/extensions/platformio.platformio-ide-2.0.1/node_modules/platformio-node-helpers/dist/index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:223:5)
	at Pipe.<anonymous> (net.js:664:12)
  ```

ivankravets, 2020-09-16:
Could you try to run this command?

```
python3 -m venv /tmp/testpyvenv
```

panciunio, 2020-09-16:
Hi,

There is empty output when I rune the above command:

```
pancio@desktop:~$ python3 -m venv /tmp/testpyvenv
pancio@desktop:~
```$ 

I checked it under running Platform.io and in idde system.

ivankravets, 2020-09-16:
Yes, something going wrong with your Python. Could you also install `apt install python3-venv`?

panciunio, 2020-09-16:
`pancio@desktop:~$ sudo apt install python3-venv
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-venv is already the newest version (3.8.2-0ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
pancio@desktop:~$ `

Whole envt. works fine. There is no issues with Python e.g. I'm using Inkscape, Dia or esptool which are Python's infrequently.

panciunio, 2020-09-17:
Hi,

I investigated deeper and I found solution. The problem was generated because Python version used by OS was lower than installed. To fix the problem in Debian way distro you should change Python version to the latest installed by using 'update-alternatives' and relogin user:
########################################
Change python version:

ls /usr/bin/python*

update-alternatives --install /usr/bin/python python /usr/bin/python2.7 **1**
update-alternatives --install /usr/bin/python python /usr/bin/python3.5 **2**
update-alternatives --install /usr/bin/python python /usr/bin/pythonLATEST **2\3**

update-alternatives --config python

#######################################

Regards,
pancio

closed 2020-09-17 by panciunio