platformio/platformio-core-installer #106: Installation Manager

thongdt89, 2020-09-16:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.0.1
  System: Windows_NT, 10.0.17134, x64

  %23 Exception
  ```
  Error:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1051)'))': /simple/platformio/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1051)'))': /simple/platformio/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1051)'))': /simple/platformio/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1051)'))': /simple/platformio/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1051)'))': /simple/platformio/
  Could not find a version that satisfies the requirement platformio (from versions: )
No matching distribution found for platformio
Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\thongdt2\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 0.3.4
Platform: Windows-10
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\Users\thongdt2\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\thongdt2\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Could not fetch URL https://pypi.org/simple/platformio/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/platformio/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1051)'))) - skipping
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1051)'))) - skipping

	at c:\Users\thongdt2\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4911
	at ChildProcess.c (c:\Users\thongdt2\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\thongdt2\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-09-16:
Do you have antivirus in a system? Could you remove `C:\Users\thongdt2\.platformio\penv` folder and try again?

ivankravets, 2020-10-16:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/152

closed 2020-10-16 by ivankravets