platformio/platformio-core-installer #110: Installation Manager

cristianro99, 2020-09-16:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.0
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.18362, x64

  %23 Exception
  ```
  Error: 
	at c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:4911
	at ChildProcess.s (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-09-18:
Do you use antivirus tools?

cristianro99, 2020-09-20:
Thousands of thanks, yes, I use Comodo, which found 8 unrecognized files, which I passed as trust, and I disabled Comodo, and I closed it from the task manager, and re-install platformio, nothing , same result, that was last night.  It's just that Comodo has restarted,
Today, at Comodo, I found 3 more locked files, which I passed as trust, and I reinstalled Platformio, and it worked.
Once again thousands of thanks.
Cristian

ivankravets, 2020-09-20:
Happy coding with PlatformIO!

closed 2020-09-20 by ivankravets