platformio/platformio-core-installer #130: FileNotFoundError exception with vscode 1.49.1 during installation

f1oat, 2020-09-22:
%23 Configuration

  VSCode: 1.49.1
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.18362, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\f1oat\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>
    main()
  File "C:\Users\f1oat\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main
    bootstrap()
  File "C:\Users\f1oat\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\f1oat\.platformio\.cache\tmp\.piocore-installer-ahhj6hho\tmph9soh0m2\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\Users\f1oat\.platformio\.cache\tmp\.piocore-installer-ahhj6hho\tmph9soh0m2\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\Users\f1oat\.platformio\.cache\tmp\.piocore-installer-ahhj6hho\tmph9soh0m2\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\Users\f1oat\.platformio\.cache\tmp\.piocore-installer-ahhj6hho\tmph9soh0m2\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\Users\f1oat\.platformio\.cache\tmp\.piocore-installer-ahhj6hho\tmph9soh0m2\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\Users\f1oat\.platformio\.cache\tmp\.piocore-installer-ahhj6hho\tmph9soh0m2\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\Users\f1oat\.platformio\.cache\tmp\.piocore-installer-ahhj6hho\tmph9soh0m2\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\Users\f1oat\.platformio\.cache\tmp\.piocore-installer-ahhj6hho\tmph9soh0m2\pioinstaller.zip\pioinstaller\__main__.py", line 52, in cli
  File "C:\Users\f1oat\.platformio\.cache\tmp\.piocore-installer-ahhj6hho\tmph9soh0m2\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core
  File "C:\Users\f1oat\.platformio\.cache\tmp\.piocore-installer-ahhj6hho\tmph9soh0m2\pioinstaller.zip\pioinstaller\penv.py", line 75, in create_core_penv
  File "C:\Users\f1oat\.platformio\.cache\tmp\.piocore-installer-ahhj6hho\tmph9soh0m2\pioinstaller.zip\pioinstaller\penv.py", line 144, in add_state_info
  File "C:\Users\f1oat\.platformio\python3\lib\subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "C:\Users\f1oat\.platformio\python3\lib\subprocess.py", line 488, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\f1oat\.platformio\python3\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Users\f1oat\.platformio\python3\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] Le fichier sp�cifi� est introuvable
 -> Installer version: 0.3.5
Platform: Windows-10
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\Users\f1oat\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\f1oat\.platformio\penv

	at c:\Users\f1oat\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:4911
	at ChildProcess.s (c:\Users\f1oat\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\f1oat\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

f1oat, 2020-09-22:
Same problem with VSCode: 1.48.2

f1oat, 2020-11-25:
Problem solved with the command ".platformio\python3\python.exe ..platformio.cache\tmp\get-platformio-0.3.5.py"
As described in #154
Thank you!

closed 2020-11-25 by f1oat

wrgallo, 2021-01-27:
> Problem solved with the command ".platformio\python3\python.exe ..platformio.cache\tmp\get-platformio-0.3.5.py"
> As described in #154
> Thank you!

That did **not** worked for me, installing [GIT](https://git-scm.com/) and restarting VSCode worked.