platformio/platformio-core-installer #152: Installation Manager

Seng22, 2020-09-27:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.2
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error:   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000029FB8FEA188>: Failed to establish a new connection: [WinError 10061] No connection could be made because the target machine actively refused it')': /simple/platformio/
  WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000029FB8FEA6C8>: Failed to establish a new connection: [WinError 10061] No connection could be made because the target machine actively refused it')': /simple/platformio/
  WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000029FB8FE5F88>: Failed to establish a new connection: [WinError 10061] No connection could be made because the target machine actively refused it')': /simple/platformio/
  WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000029FB8FE5608>: Failed to establish a new connection: [WinError 10061] No connection could be made because the target machine actively refused it')': /simple/platformio/
  WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000029FB8FE5988>: Failed to establish a new connection: [WinError 10061] No connection could be made because the target machine actively refused it')': /simple/platformio/
  ERROR: Could not find a version that satisfies the requirement platformio (from versions: none)
ERROR: No matching distribution found for platformio
Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\ISIS\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 0.3.5
Platform: Windows-10
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\Users\ISIS\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\ISIS\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio

	at c:\Users\ISIS\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:4911
	at ChildProcess.s (c:\Users\ISIS\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\ISIS\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

kqkyler, 2020-12-04:
I'm having the same issue.

FulvioSpelta, 2020-12-06:
Hi all, I'd like to share a solution that simply worked for my configuration.

**My installation problem recap**
During Build command i get "CERTIFICATE_VERIFY_FAILED" error downloading a reuqested package.

**Diagnose**
My company use a remote proxy that act as a "man in the middle" and can change the SSL certificate used to secure the connection to its own certificate. This need that the python env needs to know and validate all the certification path for this certificate. In my case the "middle" certificate comes from "blue coat systems" and is issued fom "Cloud Services CA".

For completeness this is the full path used by pip to download package:
https://api.registry.ns1.platformio.org/v3/packages?query=type%3A%22platform%22+name%3A%22espressif8266%22

**Solution**
Obviously my company have to setup our pc to be full functional, this means that in the windows cetificate repository are installed all the certificates we need to have succesfully internet surfing. Searching a bit. I've found this script:
[https://pypi.org/project/python-certifi-win32/](https://pypi.org/project/python-certifi-win32/)
that merge the certificates in the windows repository to the certificates used by pip

I've simply installed the scipt using the command shown at the project's page.
thanks to the merging the ssl verification is now ok.

I post it because it's a typical configuration for corporate users.

derwentx, 2020-12-23:
I haven't found a solution that works for my macOS setup yet, will edit this if I do, but something that might point people in the right direction is that I feel like Brew updating openSSL did this.

SrGeek, 2021-02-28:
I had a firewall.  I block all countries outside the US so I added the platformio.org website to my white list and that solved it for me.

PowerShellNibbixwoud, 2021-10-30:
Disabling the HTTPS scanning feature of my antivirus has fixed this problem for me. I suspect it's because the AV performs a man-in-the-middle attack on all HTTPS traffic in order to scan it.

karl520, 2023-10-26:
国内可以换一下国内镜像源解决
比如：
pip config set global.index-url https://pypi.tuna.tsinghua.edu.cn/simple

huster-songtao, 2023-10-27:
> 国内可以换一下国内镜像源解决 比如： `pip config set global.index-url https://pypi.tuna.tsinghua.edu.cn/simple`

PS C:\Users\tao> pip config set global.index-url https://pypi.tuna.tsinghua.edu.cn/simple
Writing to C:\Users\tao\AppData\Roaming\pip\pip.ini

Many thanks.

fpk1, 2023-10-29:
> 国内可以换一下国内镜像源解决 比如： `pip config set global.index-url https://pypi.tuna.tsinghua.edu.cn/simple`

我试了这个方法，可以

jpsaila, 2024-01-16:
This sends me to this page but i haven't found a working solution
[image]

FulvioSpelta, 2024-01-16:
> This sends me to this page but i haven't found a working solution

You posted too few info.
Could be that your problem is the "man in the middle" that use a certificate with weak key but you have to analyze more your environment.

jpsaila, 2024-02-07:
> > This sends me to this page but i haven't found a working solution
> 
> You posted too few info. Could be that your problem is the "man in the middle" that use a certificate with weak key but you have to analyze more your environment.

there's not much i can post - i got an error when installing the platformio extension on visual studio code, there was a link to follow that got me here and no working solution!

mccahan, 2024-03-23:
I was getting the same error message as @jpsaila, I was ultimately able to get it working by checking VSCode dev tools and looking for the Python command it was running, and running it in a terminal directly.

The command I had was: `/Users/[MYUSERNAME]/.platformio/.cache/tmp/get-platformio-1.2.2.py`

327w, 2024-06-24:
我发现如果出现这个错误，你可以用杀毒软件自带的网络修复功能扫描一下。我的情况就是出现了异常，并且我用软件修复了它，然后一切都解决了。

emaines, 2024-08-18:
On Windows, I was using the Malwarebytes Firewall and had to temporarily disable it, then add a filter rule, to allow PlatformIO to install, as it was failing to show the usual, can this app have permission to get outbound connections?

Reynolds-Beta, 2024-09-27:
This same error message is occurring to me as well, in this case the failure seems to be caused by my firewall which considers pypi.org to be a malicious site - not totally incorrect given the focus hackers have on the Python infrastructure even recently.

panki27, 2024-12-19:
There should be an option `--ignore-ssl` in the `get-platformio.py` installer.

Installing packages with pip through SSL decryption is possible [if the right flags are supplied](https://stackoverflow.com/a/29751768).

sunshineinwater, 2025-02-13:
> This sends me to this page but i haven't found a working solution [image]

same issue, it works fine at least 2 years before

FulvioSpelta, 2025-02-13:
> > This sends me to this page but i haven't found a working solution [image]
> 
> same issue, it works fine at least 2 years before

What is changed in your connectivity ?

clysto, 2025-03-11:
for me: use `pip config set global.trusted-host "pypi.org files.pythonhosted.org"`

J-F-Sun, 2025-04-20:
> 国内可以换一下国内镜像源解决 比如： `pip config set global.index-url https://pypi.tuna.tsinghua.edu.cn/simple`

谢谢，亲测有用。如果设置过proxy记得关一下，应该就可以了

dennisdebel, 2025-04-30:
> > 国内可以换一下国内镜像源解决 比如： `pip config set global.index-url https://pypi.tuna.tsinghua.edu.cn/simple`
> 
> 谢谢，亲测有用。如果设置过proxy记得关一下，应该就可以了

For people in China, this works indeed (Just in macos terminal), 谢谢你！

j710328466, 2025-07-09:
总结一下，需要设置python环境的两个配置：
1. pip config set global.index-url https://pypi.tuna.tsinghua.edu.cn/simple  // 用来指定国内镜像源
2. trusted-host = pypi.tuna.tsinghua.edu.cn  // 如果设置 1 还是无效就设置这个，目的是跳过 ssl 校验（生产环境慎用）

ZhouYung, 2025-07-09:
Modify the hosts file and add the domain mapping: 167.82.12.223 [files.pythonhosted.org](https://files.pythonhosted.org/). This can solve the problem, but there may still be failures in loading other resources. It's better to switch to another network.