platformio/platformio-core-installer #153: Installation Manager

Hypnos3, 2020-09-27:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.2
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.18362, x64

  %23 Exception
  ```
  Error:   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x000001FC6F695B20>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/
  WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x000001FC6F695CA0>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/
  WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x000001FC6F6742B0>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/
  WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x000001FC6F674EB0>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/
  WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x000001FC6F6746D0>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/
  ERROR: Could not find a version that satisfies the requirement platformio (from versions: none)
ERROR: No matching distribution found for platformio
Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\post\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 0.3.5
Platform: Windows-10
Python version: 3.8.2 (tags/v3.8.2:7b3ab59, Feb 25 2020, 23:03:10) [MSC v.1916 64 bit (AMD64)]
Python path: C:\Python38\python.exe
Creating a virtual environment at C:\Users\post\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio

    at c:\Users\post\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:138:23
    at ChildProcess.s (c:\Users\post\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:124:5)
    at ChildProcess.emit (events.js:223:5)
    at ChildProcess.cp.emit (c:\Users\post\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1021:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

Hypnos3, 2020-09-27:
A reason could be a missing internet connection. To thus an error message would be better than to throw this exception.

ivankravets, 2020-10-16:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/152

closed 2020-10-16 by ivankravets