platformio/platformio-core-installer #163: Installation Manager/ platformIO: distutils vs distutils-extra

UweBonnes, 2020-09-29:
%23 Description of problem
  This is on OpenSuse 15.2. Both python2-distutils-extra and python3-distutils-extra are installed. Installing platformio fails with missing python3-distutils.

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.2
  PIO IDE: v2.1.0
  System: Linux, 5.3.18-lp152.41-default, x64

  %23 Exception
  ```
  Error: Error: Can not install PlatformIO Core due to a missed `distutils` package in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-distutils

(MAY require administrator access `sudo`)
 -> Installer version: 0.3.5
Platform: Linux-5.3.18-lp152.41-default-x86_64-with-glibc2.3.4
Python version: 3.6.10 (default, Jan 16 2020, 09:12:04) [GCC]
Python path: /usr/bin/python3
Creating a virtual environment at /home/bon/.platformio/penv

	at /home/bon/.vscode/extensions/platformio.platformio-ide-2.1.0/node_modules/platformio-node-helpers/dist/index.js:1:4911
	at ChildProcess.s (/home/bon/.vscode/extensions/platformio.platformio-ide-2.1.0/node_modules/platformio-node-helpers/dist/index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

UweBonnes, 2020-09-29:
Installing as user manual with 'python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)"' and restarting code got platformio going.

ivankravets, 2020-10-16:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/85

closed 2020-10-16 by ivankravets