platformio/platformio-core-installer #190: Installation Manager

dferruzzo, 2020-10-06:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.49.3
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\dferr\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>
    main()
  File "C:\Users\dferr\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main
    bootstrap()
  File "C:\Users\dferr\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 45, in bootstrap
    import pioinstaller.__main__
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\dferr\.platformio\.cache\tmp\.piocore-installer-uus7t6qm\tmpwa8mufbu\pioinstaller.zip\pioinstaller\__init__.py", line 15, in <module>
  File "C:\Users\dferr\.conda\pkgs\python-3.8.5-he1778fa_0\Lib\logging\config.py", line 30, in <module>
    import logging.handlers
  File "C:\Users\dferr\.conda\pkgs\python-3.8.5-he1778fa_0\Lib\logging\handlers.py", line 26, in <module>
    import logging, socket, os, pickle, struct, time, re
  File "C:\Users\dferr\.conda\pkgs\python-3.8.5-he1778fa_0\Lib\socket.py", line 49, in <module>
    import _socket
ImportError: DLL load failed while importing _socket: N�o foi poss�vel encontrar o m�dulo especificado.

	at c:\Users\dferr\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:4911
	at ChildProcess.s (c:\Users\dferr\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\platformio-node-helpers\dist\index.js:1:4807)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\dferr\.vscode\extensions\platformio.platformio-ide-2.1.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2020-10-16:
> .conda

we don't support anaconda. please remove this buggy software and use the latest official Python 3.8 https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

closed 2020-10-16 by ivankravets

dferruzzo, 2020-10-16:
It worked! After a lot of work I could uninstall Anaconda/conda. The easy
part was installing python 3.8. Everything is running fine now.
Thanks a lot!

Diego Ferruzzo