platformio/platformio-core-installer #209: Gitpod error in installing

boomik305, 2020-10-13:
# Installing platform-vscode-ide 
I installing platformio-vscode-ide in gitpod from vsix file.
# Description of problem  Leave a comment...  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN  - https://github.com/platformio/platformio-vscode-ide/issues?q=is:issue   # Configuration  VSCode: 1.44.0  PIO IDE: v2.1.0  System: Linux, 4.14.174 , x64  # Exception  ```  Error: Error: Compatible PlatformIO Core not found.Reason: PlatformIO executable not found in `/home/gitpod/.platformio/penv/bin`    at c (/tmp/vscode-extensions/platformio.platformio-ide@2.1.0/extension/node_modules/platformio-node-helpers/dist/index.js:1:4911)    at ChildProcess.s (/tmp/vscode-extensions/platformio.platformio-ide@2.1.0/extension/node_modules/platformio-node-helpers/dist/index.js:1:4807)    at ChildProcess.emit (events.js:189:13)    at maybeClose (internal/child_process.js:970:16)    at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)  ```

ivankravets, 2020-10-16:
What is it?

boomik305, 2020-10-16:
This report generated PlatformIO installer.
Screenshot

[image]

closed 2020-12-22 by stale[bot]