platformio/platformio-core-installer #221: Installation Manager

reza21310, 2020-10-15:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.0
  System: Windows_NT, 10.0.18362, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\USER\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>
    main()
  File "C:\Users\USER\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main
    bootstrap()
  File "C:\Users\USER\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\USER\.platformio\.cache\tmp\.piocore-installer-f3auf236\tmplh2w1f4h\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\Users\USER\.platformio\.cache\tmp\.piocore-installer-f3auf236\tmplh2w1f4h\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\Users\USER\.platformio\.cache\tmp\.piocore-installer-f3auf236\tmplh2w1f4h\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\Users\USER\.platformio\.cache\tmp\.piocore-installer-f3auf236\tmplh2w1f4h\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\Users\USER\.platformio\.cache\tmp\.piocore-installer-f3auf236\tmplh2w1f4h\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\Users\USER\.platformio\.cache\tmp\.piocore-installer-f3auf236\tmplh2w1f4h\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\Users\USER\.platformio\.cache\tmp\.piocore-installer-f3auf236\tmplh2w1f4h\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\Users\USER\.platformio\.cache\tmp\.piocore-installer-f3auf236\tmplh2w1f4h\pioinstaller.zip\pioinstaller\__main__.py", line 52, in cli
  File "C:\Users\USER\.platformio\.cache\tmp\.piocore-installer-f3auf236\tmplh2w1f4h\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core
  File "C:\Users\USER\.platformio\.cache\tmp\.piocore-installer-f3auf236\tmplh2w1f4h\pioinstaller.zip\pioinstaller\penv.py", line 75, in create_core_penv
  File "C:\Users\USER\.platformio\.cache\tmp\.piocore-installer-f3auf236\tmplh2w1f4h\pioinstaller.zip\pioinstaller\penv.py", line 144, in add_state_info
  File "C:\Users\USER\.platformio\python3\lib\subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "C:\Users\USER\.platformio\python3\lib\subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['C:\\Users\\USER\\.platformio\\penv\\Scripts\\python.exe', '-c', "import sys

Hi i get this error when installing PlatformIO

ivankravets, 2020-10-16:
Do you use antivirus tools?

wleagle, 2020-10-20:
I have exactly the same problem with configuration:
VSCode: 1.50.1
PIO IDE: v2.1.3
System: Windows_NT x64, 10.0.19041

wleagle, 2020-10-20:
My problem solved.  It is caused by a space in my username. Space is not allowed in the file path of python.

ivankravets, 2020-10-20:
Just tried in a clean VM box and "Don Pedro" works as expected. Please destroy from your system insecure and danger software, such as dr.web, kasperskyi, avast, avg, mcafee, etc.

[image]

ambarusa, 2020-10-20:
I had no softwares installed like you mentioned, @ivankravets.

VSCode 1.50.1
Core 5.0.1, Home 3.3.0
Python 3.9 (latest)

ivankravets, 2020-10-20:
@ambarusa could you provide your output from VSCode's console? See my screenshot above. VSCode > Help > Developer tools > Console.

ambarusa, 2020-10-20:
> console.ts:137 [Extension Host] Error: Traceback (most recent call last):
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>
>     main()
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main
>     bootstrap()
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap
>     pioinstaller.__main__.main()
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\.piocore-installer-4fiaxrjr\tmp1g1qxxyl\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\.piocore-installer-4fiaxrjr\tmp1g1qxxyl\pioinstaller.zip\click\core.py", line 829, in __call__
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\.piocore-installer-4fiaxrjr\tmp1g1qxxyl\pioinstaller.zip\click\core.py", line 782, in main
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\.piocore-installer-4fiaxrjr\tmp1g1qxxyl\pioinstaller.zip\click\core.py", line 1236, in invoke
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\.piocore-installer-4fiaxrjr\tmp1g1qxxyl\pioinstaller.zip\click\core.py", line 1066, in invoke
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\.piocore-installer-4fiaxrjr\tmp1g1qxxyl\pioinstaller.zip\click\core.py", line 610, in invoke
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\.piocore-installer-4fiaxrjr\tmp1g1qxxyl\pioinstaller.zip\click\decorators.py", line 21, in new_func
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\.piocore-installer-4fiaxrjr\tmp1g1qxxyl\pioinstaller.zip\pioinstaller\__main__.py", line 52, in cli
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\.piocore-installer-4fiaxrjr\tmp1g1qxxyl\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\.piocore-installer-4fiaxrjr\tmp1g1qxxyl\pioinstaller.zip\pioinstaller\penv.py", line 75, in create_core_penv
>   File "C:\Users\Adam Ambarus\.platformio\.cache\tmp\.piocore-installer-4fiaxrjr\tmp1g1qxxyl\pioinstaller.zip\pioinstaller\penv.py", line 144, in add_state_info
>   File "C:\Users\Adam Ambarus\.platformio\python3\lib\subprocess.py", line 411, in check_output
>     **kwargs).stdout
>   File "C:\Users\Adam Ambarus\.platformio\python3\lib\subprocess.py", line 512, in run
>     output=stdout, stderr=stderr)
> subprocess.CalledProcessError: Command '['C:\\Users\\Adam Ambarus\\.platformio\\penv\\Scripts\\python.exe', '-c', "import sys; version=sys.version_info; print('%d.%d.%d'%(version[0],version[1],version[2]))"]' returned non-zero exit status 101.
>  -> Installer version: 0.3.5
> Platform: Windows-10
> Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
> Python path: C:\Users\Adam Ambarus\.platformio\python3\python.exe
> Creating a virtual environment at C:\Users\Adam Ambarus\.platformio\penv
> 	at c:\Users\Adam Ambarus\.vscode\extensions\platformio.platformio-ide-2.1.3\node_modules\platformio-node-helpers\dist\index.js:1:879089	at ChildProcess.c (c:\Users\Adam Ambarus\.vscode\extensions\platformio.platformio-ide-2.1.3\node_modules\platformio-node-helpers\dist\index.js:1:878983)	at ChildProcess.emit (events.js:223:5)	at ChildProcess.cp.emit (c:\Users\Adam Ambarus\.vscode\extensions\platformio.platformio-ide-2.1.3\node_modules\cross-spawn\lib\enoent.js:34:29)	at maybeClose (internal/child_process.js:1021:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)

ivankravets, 2020-10-20:
Could you open your CMD terminal and type?
```
"C:\Users\Adam Ambarus\.platformio\penv\Scripts\python.exe" -c "import sys; version=sys.version_info; print('%d.%d.%d'%(version[0],version[1],version[2]))"
```

ambarusa, 2020-10-20:
> Unable to create process using 'C:\Users\Adam Ambarus\.platformio\python3\python.exe -c "import sys; version=sys.version_info; print('%d.%d.%d'%(version[0],version[1],version[2]))"'

I get this response in admin mode too.

ivankravets, 2020-10-20:
> Unable to create process using

I'm not familiar with Windows OS. I used it 15 years ago last time. What is the reason that you can't call Python from the shell and I can do this from my Windows VM box? Do you use any extra software that protects you?

ambarusa, 2020-10-20:
this command works, if I'm using python.exe from ...\AppData\Local\Programs\Python\Python39

I will come with more information, after investigation.

cheintz, 2020-10-24:
@ambarusa were you able to figure this out? I'm having the same issue, except I had PlatformIO working in August, and an update broke it. Reinstalling VS Code and installing old versions of PlatformIO still do not succeed. Deleting the .platformio cache between uninstall and reinstall also does not help.

I do not have any antivirus other than Security Essentials. Disabling Security Essentials real-time protection does not allow for successful installation.

MacroYau, 2020-10-24:
Same issue here. Tried to disable anti-virus, remove `.platformio` and obsolete VS Code extension folders, but still prompts the same error. Installing the core in `penv` via `pip install -U platformio` works, though.

ivankravets, 2020-10-25:
It looks like an issue with a security policy that does not allow to run PlatformIO Core `pio.exe` that was installed via Node.JS process (VSCode uses it). Maybe, some Windows updates?

cheintz, 2020-10-26:
So some combination of uninstalling an old Python, installing a new python, and stopping a program I use called ActualWindowManager, followed by reinstalling VS Code, allowed PlatformIO Core to install. I've never had a problem with ActualWindowManager before, including installing previous versions of PlatformIO (I know I gad it working in late August). 

I've gotta say, though, the sheer number of errors people are having points to this installation procedure being super brittle. The attitude of "we don't support having Conda installed" and "you must be doing something wrong" is also worrying. I realized this is freeware, but devs dismissing installation issues for something that is designed to be easy-to-use is pretty jarring.

ambarusa, 2020-11-04:
I managed to solve the problem. I found in the logs, that it throwed an **The requested operation requires elevation - Error 740 in Windows 10**

`https://thegeekpage.com/fix-the-requested-operation-requires-elevation-error-740-in-windows-10/ `
Fix.3 and Fix.4 did the work.

ivankravets, 2020-11-04:
@ambarusa, thank you so much for the solution!!! I've just updated the initial comment.

P.S: Did you change that slider before? How did it happen?

ambarusa, 2020-11-04:
Hi Ivan,

*What do you mean by slider?* I don't know if it appeared before, or we
haven't noticed that error, but now this *Error 740* from Windows 10 OS
clearly appeared, and I just simply googled for the solution. I don't know
how important security changes they are, but now I will try to revert all
of these, and see if PIO still works. I will keep you updated anyway, and
post more on that Github issue.
Of course Ivan, you are making a huge improvement for the embedded
development, I hope I could help you with more. Usually, I use linux too,
for smart home, but I have an existent retro VFD/Nixie clock project, and
since I'm working on Windows, it's easier to develop it here. I could even
send some pictures, just to cheer you up :).
I'm an embedded developer in the automotive industry, don't know how useful
it is in this case, but I would like to support you in any way I can.

Regards,
Ádám

MacroYau, 2020-11-05:
> 
> 
> I managed to solve the problem. I found in the logs, that it throwed an **The requested operation requires elevation - Error 740 in Windows 10**
> 
> `https://thegeekpage.com/fix-the-requested-operation-requires-elevation-error-740-in-windows-10/ `
> Fix.3 and Fix.4 did the work.

Thanks @ambarusa! The same trick works for me. Specifically, I applied Fix 4 on the `.platformio` folder under my user folder.

ivankravets, 2020-11-05:
Thanks, @MacroYau !!! The guide has been updated in the top comment.

I still don't understand what is the reason. What is your Win 10 edition?

Could you open [PlatformIO CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) in VSCode and type?
```
pio system info
```

Please share here.

MacroYau, 2020-11-06:
@ivankravets Here's my `pio system info` output:

```
--------------------------  ----------------------------------------------------------
PlatformIO Core             5.0.2
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Macro Yau\.platformio
PlatformIO Core Executable  C:\Users\Macro Yau\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\macro yau\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       0
Tools & Toolchains          2
--------------------------  ----------------------------------------------------------
```

Hope it helps tracing down the issue. Thanks!

ivankravets, 2020-11-06:
@MacroYau   What is your [Win 10 edition](https://en.wikipedia.org/wiki/Windows_10_editions)?

KEdwards79, 2020-11-07:
> I managed to solve the problem. I found in the logs, that it throwed an **The requested operation requires elevation - Error 740 in Windows 10**
> 
> `https://thegeekpage.com/fix-the-requested-operation-requires-elevation-error-740-in-windows-10/ `
> Fix.3 and Fix.4 did the work.

What do I need to do (fix.3 and fix.4)?  I'm a total noob at this

ivankravets, 2020-11-07:
Please scroll up ⬆️ to the beginning. There is a solution for this issue.

KEdwards79, 2020-11-07:
I followed the  "properties ->advanced, etc...) but still get the same error.  platformio will not install, says platformio is not recognized when I try to update all
[image]

KEdwards79, 2020-11-07:
[image]
and this when I first open VSC

KEdwards79, 2020-11-12:
I have now tried all 4 fixes and still get the same error

ambarusa, 2020-11-12:
Are you running VSCode as Administrator?

KEdwards79, 2020-11-13:
yep, even tried disabling my antivirus.  It was working, then the other day I went to install a couple of recommended extensions and  after that, Platform IO IDE would not load.  Tried uninstalling VSC, the settings folder and the extensions folder, did a clean install, same results.  About to just give up at this point lol

brandesjj, 2020-11-15:
I have the same problem, nothing of the above works for me unfortunately.

MacroYau, 2020-11-16:
> 
> 
> @MacroYau What is your [Win 10 edition](https://en.wikipedia.org/wiki/Windows_10_editions)?

I am on Windows 10 Enterprise, version 2004 (build 19041.630).

flybob8888, 2020-11-23:
I get stuck here as well.
Did the fix in the first post without any luck.
No antivirus installed, no firewalls enabled.
Went through all the fixes in the thread without any luck.

Error: Traceback (most recent call last):
  File "C:\Users\Lilla Familjen\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>
    main()
  File "C:\Users\Lilla Familjen\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main
    bootstrap()
  File "C:\Users\Lilla Familjen\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\Lilla Familjen\.platformio\.cache\tmp\.piocore-installer-h5xqd7ce\tmpm2pg3h55\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\Users\Lilla Familjen\.platformio\.cache\tmp\.piocore-installer-h5xqd7ce\tmpm2pg3h55\pioinstaller.zip\click\core.py", li...

However, I just fixed it by running the python script by itself.
Navigated to C:\Users\xxxxx\.platformio\.cache\tmp
Executed get-platformio-0.3.5.py
All done.

Mikegyver, 2020-12-02:
> Please scroll up ⬆️ to the beginning. There is a solution for this issue.

hi @ivankravets . i did everything from beginning.

1st. I ticked replace all child object permission entries with inheritable permission entries from this object in the Advanced Security Setting.
2nd. I turn off / disable the antivirus
3rd. I did Fix no 2,3 & 4 for Error 740 in Windows 10 based on the tutorial https://thegeekpage.com/fix-the-requested-operation-requires-elevation-error-740-in-windows-10/

FYI. I'm using avast free antivirus & I'm running on windows 10 home edition version 20H2 (OS build 19042.662)

i do hope these information is helpful & adequate for you & your team to troubleshoot / rectify this issue. and hopefully i could get using platformIO sooner coz i'm hitting the dateline for the development. regards.

ivankravets, 2020-12-02:
> 4. I have uninstall platformIO and remove / delete %HOME_DIR%/.platformio folder and restart the computer then install back platformIO on vscode but still the errors still occur.

If you removed the folder, you need to set up permissions again?

Mikegyver, 2020-12-04:
> 
> 
> > 1. I have uninstall platformIO and remove / delete %HOME_DIR%/.platformio folder and restart the computer then install back platformIO on vscode but still the errors still occur.
> 
> If you removed the folder, you need to set up permissions again?

hi @ivankravets thanks for replying my commented. i appreciate it.

by the way, i'm confirmed after done step no 4, redid "ticked replace all child object permission entries with inheritable permission entries from this object in the Advanced Security Setting."

but unfortunately the problem still happen.

Mikegyver, 2020-12-10:
> > 1. I have uninstall platformIO and remove / delete %HOME_DIR%/.platformio folder and restart the computer then install back platformIO on vscode but still the errors still occur.
> 
> If you removed the folder, you need to set up permissions again?

Hi @ivankravets. it had been 5 days since my last reply. any update / news regarding this issue? regards.

cjwinty, 2020-12-10:
I also have the same issues, removed every thing and tried all solutions but still unable to install

Error: Traceback (most recent call last):
  File "C:\Users\Colin Winterborne\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>
    main()
  File "C:\Users\Colin Winterborne\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main
    bootstrap()
  File "C:\Users\Colin Winterborne\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\Colin Winterborne\.platformio\.cache\tmp\.piocore-installer-qbpmrf5v\tmpb_kbhmae\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\Users\Colin Winterborne\.platformio\.cache\tmp\.piocore-installer-qbpmrf5v\tmpb_kbhmae\pioinstaller.zip\cli...

pavelpetrovic, 2020-12-17:
Is it really that bad that this cannot be fixed for so long? Fresh download of visual studio code today, install PlatformIO IDE extension => error. Does it have anything to do with the fact that instead of being written in a programming language it is written in Python? 

The "solution" in the first post is not a solution. I have a space in a user name, but the .platformio folder is being created not in the ROOT folder, but in my user home folder. So where is the solution? What should I do? It's just broken. :(

```
Installing PlatformIO IDE...
It may take a few minutes depending on your connection speed
Please do not close this window and do not open other folders until this process is completed.

Debugging information is available via VSCode > Help > Toggle Developer Tools > Console.
Failed to install PlatformIO IDE.

```
`Error: Traceback (most recent call last): File "C:\Users\Pavel Petrovic\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module> main() File "C:\Users\Pavel Petrovic\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main bootstrap() File "C:\Users\Pavel Petrovic\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap pioinstaller.__main__.main() File "C:\Users\Pavel Petrovic\.platformio\.cache\tmp\.piocore-installer-o3_byxdm\tmpvsnpb20v\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main File "C:\Users\Pavel Petrovic\.platformio\.cache\tmp\.piocore-installer-o3_byxdm\tmpvsnpb20v\pioinstaller.zip\click\core.py", li...`

And what does this mean? Is it installed or is it not installed? Can it be fixed manually somehow, or the users with a space in the user name are not allowed to use PlatformIO?

ivankravets, 2020-12-22:
Could someone provide a remote session to debug this issue? Please contact me personally at me@ikravets.com.

Thanks!

Mikegyver, 2020-12-23:
> 
> 
> Could someone provide a remote session to debug this issue? Please contact me personally at [me@ikravets.com](mailto:me@ikravets.com).
> 
> Thanks!

Hi @ivankravets . thanks for your time. I'd really appreaciate your reply. Here is the debug console screenshot.

[image]

and i already contact your for the remote session to debug this issue. best regards.

Black6spdZ, 2021-01-08:
fixes here DON'T work! Win 10 x64 1909

Error: Traceback (most recent call last):
  File "C:\Users\Windows User\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>
    main()
  File "C:\Users\Windows User\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main
    bootstrap()
  File "C:\Users\Windows User\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\Windows User\.platformio\.cache\tmp\.piocore-installer-nxosfoza\tmp_t1qxrk0\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\Users\Windows User\.platformio\.cache\tmp\.piocore-installer-nxosfoza\tmp_t1qxrk0\pioinstaller.zip\click\core.py", line 829, in...

ivankravets, 2021-01-08:
Could you try to create `.platformio` empty folder in the root of disk `C:`? For example, `C:\.platformio`.

Restart VSCode. Did it help?

Black6spdZ, 2021-01-08:
yea I tried that a few times but it never worked... I logged in as another user and that did.. so something else profile related had it all hosed up

ivankravets, 2021-01-08:
@Black6spdZ what did you try "few times"? Have you read my message?

Black6spdZ, 2021-01-08:
sorry, I had only deleted the .platformio in the profile.. it was recreated again after starting the platformio-ide install but got same failure. It is working now after creating a new user

GurenWolf, 2021-01-10:
> 
> 
> I get stuck here as well.
> Did the fix in the first post without any luck.
> No antivirus installed, no firewalls enabled.
> Went through all the fixes in the thread without any luck.
> 
> Error: Traceback (most recent call last):
> File "C:\Users\Lilla Familjen.platformio.cache\tmp\get-platformio-0.3.5.py", line 69, in
> main()
> File "C:\Users\Lilla Familjen.platformio.cache\tmp\get-platformio-0.3.5.py", line 61, in main
> bootstrap()
> File "C:\Users\Lilla Familjen.platformio.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap
> pioinstaller.**main**.main()
> File "C:\Users\Lilla Familjen.platformio.cache\tmp.piocore-installer-h5xqd7ce\tmpm2pg3h55\pioinstaller.zip\pioinstaller__main__.py", line 122, in main
> File "C:\Users\Lilla Familjen.platformio.cache\tmp.piocore-installer-h5xqd7ce\tmpm2pg3h55\pioinstaller.zip\click\core.py", li...
> 
> However, I just fixed it by running the python script by itself.
> Navigated to C:\Users\xxxxx.platformio.cache\tmp
> Executed get-platformio-0.3.5.py
> All done.

This fixed the problem for me, even after get the error message on visual studio code.
Thanks a lot, I was struggling with this for months.

skylabby, 2021-01-12:
hello everyone , i have this same issue, what i did was copy my .platformio to the root, the install completed , however i have to start platformio from the icon, it doesn't autostart even though that option is selected. Mine is showing Core 5.0.4 /Home 3.3.1.

ivankravets, 2021-01-12:
@skylabby does your username contain a space?

skylabby, 2021-01-12:
> @skylabby does your username contain a space?

yes it does the default path is  c:\users\Brandon St.Rose\

Black6spdZ, 2021-01-12:
> @skylabby does your username contain a space?

this was never a problem in the past, why now?

skylabby, 2021-01-12:
> > @skylabby does your username contain a space?
> 
> this was never a problem in the past, why now?

i wondered the same thing, i had the issue for quite a while, i just gave up using the software, then today i return and went to the end of this thread and tried a couple things, and here i am again.

ivankravets, 2021-01-12:
Sorry for the issue with space. We will investigate it. We didn't change our source code of the installer. It looks there is a problem with Python Virtual Environment on Windows. In any case, we will work on the solution for the issue.

CarlosA1804, 2021-01-16:
i have a new solution, only remove the spaces of your windows user folder  https://www.youtube.com/watch?v=aiDhUfnXqCs

GurenWolf, 2021-01-16:
> 
> 
> i have a new solution, only remove the spaces of your windows user folder https://www.youtube.com/watch?v=aiDhUfnXqCs

Don't dare doing this. First of all, it requires to fiddle with computer registers, and going through all of that trouble is not even something considerable for a workaround, plus, if a minor mistake is done, it can lead to bigger issues on your machine.

Second, changing the user folder is not a good idea even with registry modifications, it can lead to incompatibilities with any other software you are using.

It is something what has to be handled on the programming language side and visual studio code with its extensions.

Definitely try the next, if it doesn't work, then we will have to wait for a better fix, or, create a new user on your machine with no spaces in its name, but don't change your user folder name:

> 
> > **However, I just fixed it by running the python script by itself.**
> > Navigated to C:\Users\xxxxx.platformio.cache\tmp
> > Executed get-platformio-0.3.5.py
> > All done.
> 
> This fixed the problem for me, even after getting the error message on visual studio code.
> Thanks a lot, I was struggling with this for months.

david-beaudette, 2021-03-05:
I also had this bug when installing platform.io in VSCode. To solve it, I first uninstalled my own installation of the latest Python3 but it did not work. I have then found that there were "aliases" defined for certain programs in Win10:

https://www.windowscentral.com/how-manage-app-execution-aliases-windows-10

There was one assigned to python3 so I disabled it. Success! Hope this can help others.

osyazi, 2021-03-05:
> También tuve este error al instalar platform.io en VSCode. Para resolverlo, primero desinstalé mi propia instalación del último Python3, pero no funcionó. Luego descubrí que había "alias" definidos para ciertos programas en Win10:
> 
> https://www.windowscentral.com/how-manage-app-execution-aliases-windows-10
> 
> Había uno asignado a python3, así que lo desactivé. ¡Éxito! Espero que esto pueda ayudar a otros. Tenga en cuenta que instalé desde la línea de comando usando:
> 
> C: \ Users [USER] .platformio> python3 \ python.exe .cache \ tmp \ get-platformio-1.0.0.py

Muchas gracias, me acabas de solucionar el problema a mi también!

ReneMtzOl, 2021-03-05:
Nothing worked for me, then I search a little and found that it could be python virtual environment related, so I checked what version PlatformIO was downloading and it was 3.9, so I downloaded Python 3.6.6 (no particular reason) and executed the file "get-platformio-1.0.0.py" on my own command line and it worked fine.

PabloSoligo2014, 2021-03-05:
I have finally solved the problem. As @osyazi said, windows users, please disable "the aliases"... 
[image]

cestmirp, 2021-03-06:
Hi.
Wher is a problem in my instalation :-/
:

Error: Traceback (most recent call last):
  File "C:\.platformio\.cache\tmp\.piocore-installer-0j6h2e05\tmpxgoplbjy\pioinstaller.zip\pioinstaller\python.py", line 170, in find_compatible_pythons
  File "C:\.platformio\python3\lib\subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\.platformio\python3\lib\subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\uzivatel\\AppData\\Local\\Microsoft\\WindowsApps\\python3.exe', 'C:\\.platformio\\.cache\\tmp\\get-platformio-1.0.0.py', '--no-shutdown-piohome', 'check', 'python']' returned non-z...

webdany2k, 2021-03-06:
I got solved the issue getting on the local admin account on windows 10, downloaded System Installer 64 bits from https://code.visualstudio.com/download then after the setup i was able to install PlatformIO without any issues on Admin account, i got back to my user and now PlatformIO got installed without issues too.... finally the nightmare ends!!!

[53 more replies]