platformio/platformio-core-installer #227: Installation Manager

ahanbose18, 2020-10-17:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.2
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>
    main()
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main
    bootstrap()
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\.piocore-installer-ws4b3n45\tmpvqbantzu\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\.piocore-installer-ws4b3n45\tmpvqbantzu\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\.piocore-installer-ws4b3n45\tmpvqbantzu\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\.piocore-installer-ws4b3n45\tmpvqbantzu\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\.piocore-installer-ws4b3n45\tmpvqbantzu\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\.piocore-installer-ws4b3n45\tmpvqbantzu\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\.piocore-installer-ws4b3n45\tmpvqbantzu\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\.piocore-installer-ws4b3n45\tmpvqbantzu\pioinstaller.zip\pioinstaller\__main__.py", line 52, in cli
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\.piocore-installer-ws4b3n45\tmpvqbantzu\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\.piocore-installer-ws4b3n45\tmpvqbantzu\pioinstaller.zip\pioinstaller\penv.py", line 75, in create_core_penv
  File "C:\Users\Ahan Bose\.platformio\.cache\tmp\.piocore-installer-ws4b3n45\tmpvqbantzu\pioinstaller.zip\pioinstaller\penv.py", line 144, in add_state_info
  File "C:\Users\Ahan Bose\.platformio\python3\lib\subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "C:\Users\Ahan Bose\.platformio\python3\lib\subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['C:\\Users\\Ahan Bose\\.platformio\\penv\\Scripts\\python.exe', '-c', "import sys

ahanbose18, 2020-10-17:
Unable to install Platform IO on VSCODE

sebby49, 2020-10-19:
i get the same error over and over again and nobody seems to know what the problem is its all gibberish to me i just want a new marlinfirmware with bltouch why is it so fucking hard to just post a compiled version !!!!

ivankravets, 2020-10-20:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/221

closed 2020-10-20 by ivankravets