platformio/platformio-core-installer #232: Installation Manager

sebby49, 2020-10-19:
%23 Description of problem
 it keeps doing this and it will not instal platformIO IDE

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.1.3
  System: Windows_NT, 6.1.7600, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>
    main()
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main
    bootstrap()
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\.piocore-installer-tlfunted\tmp0do50vey\pioinstaller.zip\pioinstaller\__main__.py", line 122, in main
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\.piocore-installer-tlfunted\tmp0do50vey\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\.piocore-installer-tlfunted\tmp0do50vey\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\.piocore-installer-tlfunted\tmp0do50vey\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\.piocore-installer-tlfunted\tmp0do50vey\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\.piocore-installer-tlfunted\tmp0do50vey\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\.piocore-installer-tlfunted\tmp0do50vey\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\.piocore-installer-tlfunted\tmp0do50vey\pioinstaller.zip\pioinstaller\__main__.py", line 52, in cli
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\.piocore-installer-tlfunted\tmp0do50vey\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\.piocore-installer-tlfunted\tmp0do50vey\pioinstaller.zip\pioinstaller\penv.py", line 75, in create_core_penv
  File "C:\Users\bas van swelm\.platformio\.cache\tmp\.piocore-installer-tlfunted\tmp0do50vey\pioinstaller.zip\pioinstaller\penv.py", line 144, in add_state_info
  File "C:\Users\bas van swelm\.platformio\python3\lib\subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "C:\Users\bas van swelm\.platformio\python3\lib\subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['C:\\Users\\bas van swelm\\.platformio\\penv\\Scripts\\python.exe', '-c', "import sys

ivankravets, 2020-10-20:
Do you use antivirus tools?

ivankravets, 2020-10-20:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/221

closed 2020-10-20 by ivankravets