platformio/platformio-core-installer #324: Installation Manager

scorpio66666, 2020-11-17:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.51.1
  PIO IDE: v2.2.1
  System: Darwin, 15.6.0, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.5
Platform: Darwin-15.6.0
Python version: 2.7.10 (default, Oct 23 2015, 19:19:21) 
[GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.59.5)]
Python path: /usr/bin/python
Creating a virtual environment at /Users/scottgoodman/.platformio/penv

	at /Users/scottgoodman/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275
	at ChildProcess.c (/Users/scottgoodman/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

didiergm, 2020-11-17:
exact same issue - except environment details: 
System: Linux, 5.4.0-54-generic, x64
Platform: Linux-5.4.0-54-generic-x86_64-with-glibc2.29
Python version: 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0]
Python path: /usr/bin/python3

**It happened after the update to vscode 1.51.1** -

ivankravets, 2021-01-21:
Please install the latest Python 3 as described in https://docs.platformio.org/en/latest/faq.html#install-python-interpreter and restart VSCode.

closed 2021-01-21 by ivankravets