platformio/platformio-core-installer #356: Installation Manager

fcoury, 2020-11-28:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.51.1
  PIO IDE: v2.2.1
  System: Darwin, 19.5.0, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/Users/fcoury/.platformio/python3/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Users/fcoury/.platformio/python3/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/fcoury/.platformio/penv/lib/python3.8/site-packages/pip/__main__.py", line 23, in <module>
    from pip._internal.cli.main import main as _main  %23 isort:skip %23 noqa
  File "/Users/fcoury/.platformio/penv/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 10, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/Users/fcoury/.platformio/penv/lib/python3.8/site-packages/pip/_internal/cli/autocompletion.py", line 9, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/Users/fcoury/.platformio/penv/lib/python3.8/site-packages/pip/_internal/cli/main_parser.py", line 7, in <module>
    from pip._internal.cli import cmdoptions
  File "/Users/fcoury/.platformio/penv/lib/python3.8/site-packages/pip/_internal/cli/cmdoptions.py", line 23, in <module>
    from pip._internal.cli.progress_bars import BAR_TYPES
  File "/Users/fcoury/.platformio/penv/lib/python3.8/site-packages/pip/_internal/cli/progress_bars.py", line 8, in <module>
    from pip._vendor.progress.bar import Bar, FillingCirclesBar, IncrementalBar
ModuleNotFoundError: No module named 'pip._vendor.progress'
Error: Could not install PlatformIO Core: Command '['/Users/fcoury/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 0.3.5
Platform: macOS-10.15.5
Python version: 3.8.4 (default, Jul 14 2020, 15:19:59) 
[Clang 10.0.0 (clang-1000.10.44.4)]
Python path: /Users/fcoury/.platformio/python3/bin/python3
Creating a virtual environment at /Users/fcoury/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core

	at /Users/fcoury/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275
	at ChildProcess.c (/Users/fcoury/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:223:5)
	at Pipe.<anonymous> (net.js:664:12)
  ```

ivankravets, 2021-01-21:
Have you resolved this issue?

fcoury, 2021-01-21:
I honestly don't remember, I think I gave up but not sure.

ivankravets, 2021-01-21:
Please install the latest Python 3 https://docs.platformio.org/en/latest/faq.html#install-python-interpreter and restart VSCode.

closed 2021-01-21 by ivankravets