platformio/platformio-core-installer #373: Installation Manager

Auravendill, 2020-12-07:
%23 Description of problem
  It seems like plattformio has a problem with pip and or conda
  conda-env list doesn't show the environment plattformio created
  pip is installed in my base environment

  %23 Configuration

  VSCode: 1.51.1
  PIO IDE: v2.2.1
  System: Linux, 4.19.0-13-amd64, x64

  %23 Exception
  ```
  Error: /home/daniel/.platformio/penv/bin/python: No module named pip
Error: Could not install PlatformIO Core: Command '['/home/daniel/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 0.3.5
Platform: Linux-4.19.0-13-amd64-x86_64-with-glibc2.10
Python version: 3.7.8 | packaged by conda-forge | (default, Nov 17 2020, 23:45:15) 
[GCC 7.5.0]
Python path: /home/daniel/Programme/miniconda3/bin/python3
Creating a virtual environment at /home/daniel/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core

    at cmd (/home/daniel/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:148:25)
    at ChildProcess.c (/home/daniel/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:138:7)
    at ChildProcess.emit (events.js:223:5)
    at maybeClose (internal/child_process.js:1021:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

Auravendill, 2020-12-07:
Current workaround: Installing platformio manually via conda after uninstalling platformio from vs code and removing the .platformio folder in the user folder. After that installing it still failed, but at least it is usable again, since platformio is now installed and gets detected by the part of the extension that did install. I built the firmware for my CR-10 again to confirm it and it is indeed working rn.

But that is obviously not the desired expirience. Restarting VS Code will always trigger a new installation try that will always fail.

ivankravets, 2021-01-21:
Sorry, we dropped support for conda to avoid a ton of issues that we had with it. Please install the latest Python 3 and restart VSCode.

closed 2021-01-21 by ivankravets