platformio/platformio-core-installer #388: Installation Manager

xiaohaoxiang, 2020-12-16:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.0
  PIO IDE: v2.2.1
  System: Linux, 5.8.18-1-MANJARO, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/xhx/.platformio/penv/lib/python3.8/site-packages/pip/__main__.py", line 23, in <module>
    from pip._internal.cli.main import main as _main  %23 isort:skip %23 noqa
  File "/home/xhx/.platformio/penv/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 10, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/home/xhx/.platformio/penv/lib/python3.8/site-packages/pip/_internal/cli/autocompletion.py", line 9, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/home/xhx/.platformio/penv/lib/python3.8/site-packages/pip/_internal/cli/main_parser.py", line 7, in <module>
    from pip._internal.cli import cmdoptions
  File "/home/xhx/.platformio/penv/lib/python3.8/site-packages/pip/_internal/cli/cmdoptions.py", line 23, in <module>
    from pip._vendor.packaging.utils import canonicalize_name
ModuleNotFoundError: No module named 'pip._vendor.packaging'
Error: Could not install PlatformIO Core: Command '['/home/xhx/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 0.3.5
Platform: Linux-5.8.18-1-MANJARO-x86_64-with-glibc2.2.5
Python version: 3.8.6 (default, Sep 30 2020, 04:00:38) 
[GCC 10.2.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/xhx/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core

	at /home/xhx/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275
	at ChildProcess.c (/home/xhx/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2021-01-21:
Have you resolved this issue?

xiaohaoxiang, 2021-01-23:
I don't remember reporting this issue. It seems to have been reported automatically after an installation failure.
Sorry to bother you.

closed 2021-01-23 by xiaohaoxiang