platformio/platformio-core-installer #393: my pio do not show the boards list.

newbeast-simon, 2020-12-21:
It shows that I successfully installed the pio core, but when I run command "pio boards", it shows "fail to check for pio upgrades ", then I run the command again, It shows nothing. 
My system is win7, and I'm from China, and the net connection is not so good in my country, you know. 
[image]
[image]
Thank you!

newbeast-simon, 2020-12-22:
[image]
When running command "pio update",it shows like this, and nothing else happend. There is no directory "lib" there at all.

newbeast-simon, 2020-12-22:
My Clion installed pio plug.  When newing a pio project,   it shows like this and then have no response.
[image]

newbeast-simon, 2020-12-22:
Anyone help?

newbeast-simon, 2020-12-25:
I got some clue. My python environment is Anaconda, it seems that the pio-core-installer don't support Anaconda.
"pip install" is not allways work in my internet environment, on the contrary, "conda install" works well.

ivankravets, 2021-01-21:
Have you resolved this issue?

ivankravets, 2021-01-21:
yes, we don't support anaconda. Please install the latest official Python 3 https://docs.platformio.org/en/latest/faq.html#install-python-interpreter and restart installer

closed 2021-01-21 by ivankravets