platformio/platformio-core-installer #396: Installation Manager

aschierider, 2020-12-23:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\aschi\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 69, in <module>
    main()
  File "C:\Users\aschi\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 61, in main
    bootstrap()
  File "C:\Users\aschi\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 45, in bootstrap
    import pioinstaller.__main__
  File "C:\Users\aschi\.platformio\.cache\tmp\.piocore-installer-gl4gkof_\tmpsyf9hwa1\pioinstaller.zip\pioinstaller\__init__.py", line 15, in <module>
  File "C:\Users\aschi\.platformio\python3\lib\logging\config.py", line 30, in <module>
    import logging.handlers
  File "C:\Users\aschi\.platformio\python3\lib\logging\handlers.py", line 26, in <module>
    import logging, socket, os, pickle, struct, time, re
  File "C:\Users\aschi\.platformio\python3\lib\socket.py", line 49, in <module>
    import _socket
ModuleNotFoundError: No module named '_socket'

	at c:\Users\aschi\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879275
	at ChildProcess.c (c:\Users\aschi\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\aschi\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2021-01-21:
Have you resolved this issue?

aschierider, 2021-01-21:
Yes, thanx

ivankravets, 2021-01-21:
How did you solve it?

closed 2021-01-21 by ivankravets