platformio/platformio-core-installer #399: Installation Manager

prakash52kar, 2020-12-24:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.5
Platform: Windows-10
Python version: 3.7.4 (default, Aug  9 2019, 18:34:13) [MSC v.1915 64 bit (AMD64)]
Python path: C:\ProgramData\Anaconda3\python.exe
Creating a virtual environment at C:\Users\ipr\.platformio\penv

	at c:\Users\ipr\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879275
	at ChildProcess.c (c:\Users\ipr\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\ipr\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2021-01-21:
Have you resolved this issue?

prakash52kar, 2021-01-22:
No!

ivankravets, 2021-01-22:
1. Could you try to UNINSTALL all Python from the system?
2. Install the latest official Python following this guide https://docs.platformio.org/en/latest/faq.html#install-python-interpreter
3. Restart VSCode.

closed 2021-01-22 by ivankravets