platformio/platformio-core-installer #409: Installation Manager

erefour1D, 2020-12-28:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Windows_NT, 6.1.7601, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.5
Platform: Windows-7
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\Users\OneDirect\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\OneDirect\.platformio\penv

	at c:\Users\OneDirect\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879275
	at ChildProcess.c (c:\Users\OneDirect\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\OneDirect\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2021-01-21:
Have you resolved this issue?

erefour1D, 2021-01-21:
Hey,

Yes I think I resolved it by downloading another version of VS Studios. I'll close this topic.