platformio/platformio-core-installer #410: trying to install Platform-ide on a MAC

windy54, 2020-12-29:
For the last  6 to 9 months I have been trying to install the platform io ide extension for Visual Code Editor.

Visual code installs but the platform io ide does not.

I did find one mistake where I was running python 2 and not 3, but that has not fixed the same issue.
This morning I tried to install platform io ide through the atom editor and that does not work.
Have tried to google to see if anyone else has had a problem but have not been able to find an answer, that could be because I am googling the wrong question of course.

Here is the output from atom installation:

`Error: Traceback (most recent call last): File "/Users/stevengale/.platformio/.cache/tmp/get-platformio-0.3.5.py", line 69, in main() File "/Users/stevengale/.platformio/.cache/tmp/get-platformio-0.3.5.py", line 61, in main bootstrap() File "/Users/stevengale/.platformio/.cache/tmp/get-platformio-0.3.5.py", line 47, in bootstrap pioinstaller.main.main() File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-z1t5awct/tmp14lg7jtq/pioinstaller.zip/pioinstaller/main.py", line 122, in main File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-z1t5awct/tmp14lg7jtq/pioinstaller.zip/click/core.py", line 829, in call File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-z1t5awct/tmp14lg7jtq/pioinstaller.zip/click/core.py", line 782, in main File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-z1t5awct/tmp14lg7jtq/pioinstaller.zip/click/core.py", line 1236, in invoke File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-z1t5awct/tmp14lg7jtq/pioinstaller.zip/click/core.py", line 1066, in invoke File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-z1t5awct/tmp14lg7jtq/pioinstaller.zip/click/core.py", line 610, in invoke File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-z1t5awct/tmp14lg7jtq/pioinstaller.zip/click/decorators.py", line 21, in new_func File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-z1t5awct/tmp14lg7jtq/pioinstaller.zip/pioinstaller/main.py", line 52, in cli File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-z1t5awct/tmp14lg7jtq/pioinstaller.zip/pioinstaller/core.py", line 72, in install_platformio_core File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-z1t5awct/tmp14lg7jtq/pioinstaller.zip/pioinstaller/penv.py", line 52, in create_core_penv File "/Users/stevengale/.platformio/.cache/tmp/.piocore-installer-z1t5awct/tmp14lg7jtq/pioinstaller.zip/pioinstaller/python.py", line 170, in find_compatible_pythons File "/Users/stevengale/.platformio/python3/lib/python3.8/subprocess.py", line 411, in check_output return run(popenargs, stdout=PIPE, timeout=timeout, check=True, File "/Users/stevengale/.platformio/python3/lib/python3.8/subprocess.py", line 489, in run with Popen(popenargs, **kwargs) as process: File "/Users/stevengale/.platformio/python3/lib/python3.8/subprocess.py", line 854, in init self._execute_child(args, executable, preexec_fn, close_fds, File "/Users/stevengale/.platformio/python3/lib/python3.8/subprocess.py", line 1702, in _execute_child raise child_exception_type(errno_num, err_msg, err_filename) OSError: [Errno 86] Bad CPU type in executable: '/usr/local/bin/python2' -> Installer version: 0.3.5 Platform: macOS-10.16 Python version: 3.8.4 (default, Jul 14 2020, 15:19:59) [Clang 10.0.0 (clang-1000.10.44.4)] Python path: /Users/stevengale/.platformio/python3/bin/python3 Creating a virtual environment at /Users/stevengale/.platformio/penv

at /Users/stevengale/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:5004
at ChildProcess.s (/Users/stevengale/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:4900)
at ChildProcess.emit (events.js:200:13)
at maybeClose (internal/child_process.js:1021:16)
at Socket.<anonymous> (internal/child_process.js:430:11)
at Socket.emit (events.js:200:13)
at Pipe.<anonymous> (net.js:586:12)`

I think I get a similar error from visual code installation.

Any ideas as to what I can try to fix the problem.
I am wanting to use platform io for the esp-rfid project and also to follow random nerd tutorials for the ESP 32.

cheers

Steve

ivankravets, 2020-12-29:
What is your OS version?

windy54, 2020-12-29:
Sorry I meant to add that,
I have just updated to Big Sur 11.1 which is why I thought I would try it again.
I had the same issue with Catalina.

Steve

ivankravets, 2020-12-29:
```
/Users/stevengale/.atom/
```

It seems you use Atom instead of VSCode. Please use PlatformIO IDE for VSCode. Also, please remove `/Users/stevengale/.platformio/` folder and restart VSCode.

closed 2020-12-29 by ivankravets