platformio/platformio-core-installer #420: Installation Manager

fritz-net, 2021-01-02:
# Description of problem

When i start vscode the following exception apears. This is since version 2 (I tried 2.0 and 2.1), however version 1 works fine.

running it manualy in cmd works also fine
`python C:\Users\user\.platformio\.cache\tmp\get-platformio-0.3.4.py` 

# Configuration
VSCode: 1.52.1
PIO IDE: v2.0.1
System: Windows_NT, 6.1.7601, x64

# Exception

``` Error: Traceback (most recent call last):

File "C:\Users\user\.platformio\.cache\tmp\get-platformio-0.3.4.py", line 19, in <module> import shutil
File "C:\Python35-32\Lib\shutil.py", line 10, in <module> import fnmatch
File "C:\Python35-32\Lib\fnmatch.py", line 14, in <module> import re
File "C:\Python35-32\Lib\re.py", line 123, in <module> import sre_compile
File "C:\Python35-32\Lib\sre_compile.py", line 17, in <module> assert _sre.MAGIC == MAGIC, "SRE module mismatch" 
AssertionError: SRE module mismatch at args
(c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:141:25) at ChildProcess.c
(c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:132:7) at ChildProcess.emit
(events.js:223:5) at ChildProcess.cp.emit
(c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.0.1\node_modules\cross-spawn\lib\enoent.js:34:29) at maybeClose (internal/child_process.js:1021:16) at Socket.
<anonymous> (internal/child_process.js:430:11) at Socket.emit (events.js:223:5) at Pipe.<anonymous> (net.js:664:12) ```

fritz-net, 2021-01-02:
This is the exception for
VSCode: 1.52.1
PIO IDE: v2.2.1

``` Error: Traceback (most recent call last): File "C:\Users\user\.platformio\.cache\tmp\get-platformio-0.3.5.py", line 19, in <module> import shutil File "C:\Python35-32\Lib\shutil.py", line 10, in <module> import fnmatch File "C:\Python35-32\Lib\fnmatch.py", line 14, in <module> import re File "C:\Python35-32\Lib\re.py", line 123, in <module> import sre_compile File "C:\Python35-32\Lib\sre_compile.py", line 17, in <module> assert _sre.MAGIC == MAGIC, "SRE module mismatch" AssertionError: SRE module mismatch at cmd (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:148:25) at ChildProcess.c (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:138:7) at ChildProcess.emit (events.js:223:5) at ChildProcess.cp.emit (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\cross-spawn\lib\enoent.js:34:29) at maybeClose (internal/child_process.js:1021:16) at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5) ```

ivankravets, 2021-01-21:
> C:\Python35-3

Please uninstall ancient Python 3.5 and install the latest following C:\Python35-3

closed 2021-01-21 by ivankravets