platformio/platformio-core-installer #421: Installation Manager

Sherman-c, 2021-01-03:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: 
	at c:\Users\Sherman\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879275
	at ChildProcess.c (c:\Users\Sherman\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at ChildProcess.cp.emit (c:\Users\Sherman\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2021-01-21:
Have you resolved this issue?

Sherman-c, 2021-02-08:
Hi Ivan

Sorry for the late reply, no the issue still remains.

Regards
Sherman

ivankravets, 2021-02-08:
Do you use 3rd party antivirus tools in a system?

Sherman-c, 2021-02-09:
Hi Ivan,

Yes i do have antivirus software, comodo, but i tried disabled antivirus,
before installing vscode and platformio, the result were the same,  it
seems something wrong on the environment variable setup.

Regards,
Sherman

closed 2021-03-18 by stale[bot]