platformio/platformio-core-installer #423: Installation Manager

koalabi, 2021-01-03:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Linux, 5.4.0-58-generic, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 0.3.5
Platform: Linux-5.4.0-58-generic-x86_64-with-glibc2.29
Python version: 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/alain/.platformio/penv

	at /mnt/data2/homes/alain/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275
	at ChildProcess.c (/mnt/data2/homes/alain/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

koalabi, 2021-01-03:
Additional information: attempting to install PlatformIO Core independently, from the command line, on the same machine fails with the same message.
(Machine is a pristine installation of Ubuntu 20.04 (64bit)).

koalabi, 2021-01-04:
More detailed tracing information (problem is: no clear diagnostic is provided - already out of the box)

$ python3 -d -v ./get-platformio.py 
import _frozen_importlib # frozen
import _imp # builtin
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import '_warnings' # <class '_frozen_importlib.BuiltinImporter'>
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
import '_frozen_importlib_external' # <class '_frozen_importlib.FrozenImporter'>
import '_io' # <class '_frozen_importlib.BuiltinImporter'>
import 'marshal' # <class '_frozen_importlib.BuiltinImporter'>
import 'posix' # <class '_frozen_importlib.BuiltinImporter'>
import _thread # previously loaded ('_thread')
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import _weakref # previously loaded ('_weakref')
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
# installing zipimport hook
import 'time' # <class '_frozen_importlib.BuiltinImporter'>
import 'zipimport' # <class '_frozen_importlib.FrozenImporter'>
# installed zipimport hook
# /usr/lib/python3.8/encodings/__pycache__/__init__.cpython-38.pyc matches /usr/lib/python3.8/encodings/__init__.py
# code object from '/usr/lib/python3.8/encodings/__pycache__/__init__.cpython-38.pyc'
# /usr/lib/python3.8/__pycache__/codecs.cpython-38.pyc matches /usr/lib/python3.8/codecs.py
# code object from '/usr/lib/python3.8/__pycache__/codecs.cpython-38.pyc'
import '_codecs' # <class '_frozen_importlib.BuiltinImporter'>
import 'codecs' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d8ee190>
# /usr/lib/python3.8/encodings/__pycache__/aliases.cpython-38.pyc matches /usr/lib/python3.8/encodings/aliases.py
# code object from '/usr/lib/python3.8/encodings/__pycache__/aliases.cpython-38.pyc'
import 'encodings.aliases' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d8fd760>
import 'encodings' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d94ef40>
# /usr/lib/python3.8/encodings/__pycache__/utf_8.cpython-38.pyc matches /usr/lib/python3.8/encodings/utf_8.py
# code object from '/usr/lib/python3.8/encodings/__pycache__/utf_8.cpython-38.pyc'
import 'encodings.utf_8' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d94ef70>
import '_signal' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.8/encodings/__pycache__/latin_1.cpython-38.pyc matches /usr/lib/python3.8/encodings/latin_1.py
# code object from '/usr/lib/python3.8/encodings/__pycache__/latin_1.cpython-38.pyc'
import 'encodings.latin_1' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d8fd850>
# /usr/lib/python3.8/__pycache__/io.cpython-38.pyc matches /usr/lib/python3.8/io.py
# code object from '/usr/lib/python3.8/__pycache__/io.cpython-38.pyc'
# /usr/lib/python3.8/__pycache__/abc.cpython-38.pyc matches /usr/lib/python3.8/abc.py
# code object from '/usr/lib/python3.8/__pycache__/abc.cpython-38.pyc'
import '_abc' # <class '_frozen_importlib.BuiltinImporter'>
import 'abc' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d8fdd00>
import 'io' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d8fda30>
# /usr/lib/python3.8/__pycache__/site.cpython-38.pyc matches /usr/lib/python3.8/site.py
# code object from '/usr/lib/python3.8/__pycache__/site.cpython-38.pyc'
# /usr/lib/python3.8/__pycache__/os.cpython-38.pyc matches /usr/lib/python3.8/os.py
# code object from '/usr/lib/python3.8/__pycache__/os.cpython-38.pyc'
# /usr/lib/python3.8/__pycache__/stat.cpython-38.pyc matches /usr/lib/python3.8/stat.py
# code object from '/usr/lib/python3.8/__pycache__/stat.cpython-38.pyc'
import '_stat' # <class '_frozen_importlib.BuiltinImporter'>
import 'stat' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d8b2670>
# /usr/lib/python3.8/__pycache__/_collections_abc.cpython-38.pyc matches /usr/lib/python3.8/_collections_abc.py
# code object from '/usr/lib/python3.8/__pycache__/_collections_abc.cpython-38.pyc'
import '_collections_abc' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d8b2700>
# /usr/lib/python3.8/__pycache__/posixpath.cpython-38.pyc matches /usr/lib/python3.8/posixpath.py
# code object from '/usr/lib/python3.8/__pycache__/posixpath.cpython-38.pyc'
# /usr/lib/python3.8/__pycache__/genericpath.cpython-38.pyc matches /usr/lib/python3.8/genericpath.py
# code object from '/usr/lib/python3.8/__pycache__/genericpath.cpython-38.pyc'
import 'genericpath' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d8d4f40>
import 'posixpath' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d8b2790>
import 'os' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d915d90>
# /usr/lib/python3.8/__pycache__/_sitebuiltins.cpython-38.pyc matches /usr/lib/python3.8/_sitebuiltins.py
# code object from '/usr/lib/python3.8/__pycache__/_sitebuiltins.cpython-38.pyc'
import '_sitebuiltins' # <_frozen_importlib_external.SourceFileLoader object at 0x7f569d929fd0>
# /usr/lib/python3.8/__pycache__/sitecustomize.cpython-38.pyc matches /usr/lib/python3.8/sitecustomize.py
# code object from '/usr/lib/python3.8/__pycache__/sitecustomize.cpython-38.pyc'
# /usr/lib/python3/dist-packages/__pycache__/apport_python_hook.cpython-38.pyc matches /usr/lib/python3/dist-packages/apport_python_hook.py
# code object from '/usr/lib/python3/dist-packages/__pycache__/apport_python_hook.cpython-38.pyc'
[... cut ...]
# destroy marshal
# cleanup[3] wiping _io
# destroy io
# cleanup[3] wiping _warnings
# cleanup[3] wiping _imp
# destroy _imp
# cleanup[3] wiping _frozen_importlib
# cleanup[3] wiping sys
# cleanup[3] wiping builtins
# destroy stat
# destroy genericpath
# destroy _sre
# destroy sre_parse
# destroy functools
# destroy tokenize
# destroy _struct
# destroy _compat_pickle
# destroy _pickle
# destroy _heapq
# destroy socket
# destroy selectors
# destroy select
# destroy _frozen_importlib
# clear sys.audit hooks

koalabi, 2021-01-05:
(Remotely) Debugging using VSCode provides the following root cause:

<verbatim>
Installer version: 0.3.5
Platform: Linux-5.4.0-58-generic-x86_64-with-glibc2.29
Python version: 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/alain/.platformio/penv
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/mnt/data2/homes/alain/.vscode-server-insiders/extensions/ms-python.python-2020.12.424452561/pythonFiles/lib/python/debugpy/__main__.py", line 45, in <module>
    cli.main()
  File "/mnt/data2/homes/alain/.vscode-server-insiders/extensions/ms-python.python-2020.12.424452561/pythonFiles/lib/python/debugpy/../debugpy/server/cli.py", line 444, in main
    run()
  File "/mnt/data2/homes/alain/.vscode-server-insiders/extensions/ms-python.python-2020.12.424452561/pythonFiles/lib/python/debugpy/../debugpy/server/cli.py", line 285, in run_file
    runpy.run_path(target_as_str, run_name=compat.force_str("__main__"))
  File "/usr/lib/python3.8/runpy.py", line 265, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "/usr/lib/python3.8/runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/alain/devel/python/platformio-core-installer/get-platformio.py", line 68, in <module>
    main()
  File "/home/alain/devel/python/platformio-core-installer/get-platformio.py", line 60, in main
    bootstrap()
  File "/home/alain/devel/python/platformio-core-installer/get-platformio.py", line 46, in bootstrap
    pioinstaller.__main__.main()
  File "/mnt/data2/homes/alain/devel/python/platformio-core-installer/.piocore-installer-ekv_m331/tmp9z3t1ygg/pioinstaller.zip/pioinstaller/__main__.py", line 122, in main
  File "/mnt/data2/homes/alain/devel/python/platformio-core-installer/.piocore-installer-ekv_m331/tmp9z3t1ygg/pioinstaller.zip/click/core.py", line 829, in __call__
  File "/mnt/data2/homes/alain/devel/python/platformio-core-installer/.piocore-installer-ekv_m331/tmp9z3t1ygg/pioinstaller.zip/click/core.py", line 782, in main
  File "/mnt/data2/homes/alain/devel/python/platformio-core-installer/.piocore-installer-ekv_m331/tmp9z3t1ygg/pioinstaller.zip/click/core.py", line 1236, in invoke
  File "/mnt/data2/homes/alain/devel/python/platformio-core-installer/.piocore-installer-ekv_m331/tmp9z3t1ygg/pioinstaller.zip/click/core.py", line 1066, in invoke
  File "/mnt/data2/homes/alain/devel/python/platformio-core-installer/.piocore-installer-ekv_m331/tmp9z3t1ygg/pioinstaller.zip/click/core.py", line 610, in invoke
  File "/mnt/data2/homes/alain/devel/python/platformio-core-installer/.piocore-installer-ekv_m331/tmp9z3t1ygg/pioinstaller.zip/click/decorators.py", line 21, in new_func
  File "/mnt/data2/homes/alain/devel/python/platformio-core-installer/.piocore-installer-ekv_m331/tmp9z3t1ygg/pioinstaller.zip/pioinstaller/__main__.py", line 52, in cli
  File "/mnt/data2/homes/alain/devel/python/platformio-core-installer/.piocore-installer-ekv_m331/tmp9z3t1ygg/pioinstaller.zip/pioinstaller/core.py", line 72, in install_platformio_core
  File "/mnt/data2/homes/alain/devel/python/platformio-core-installer/.piocore-installer-ekv_m331/tmp9z3t1ygg/pioinstaller.zip/pioinstaller/penv.py", line 75, in create_core_penv
  File "/mnt/data2/homes/alain/devel/python/platformio-core-installer/.piocore-installer-ekv_m331/tmp9z3t1ygg/pioinstaller.zip/pioinstaller/penv.py", line 143, in add_state_info
  File "/usr/lib/python3.8/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
<b>
FileNotFoundError: [Errno 2] No such file or directory: '/home/alain/.platformio/penv/bin/python'
</b>
</verbatim>

ivankravets, 2021-01-21:
Have you resolved this issue?

koalabi, 2021-01-21:
Hi Ivan,

Thank you for asking. I must confess I don't understand very well what the problem is as - if I remember well - when I did look into ~/.platformio/penv, bin/python was there.

Do you have any troubleshooting suggestion?

Thanks in advance.

Best regards

koalabi, 2021-01-21:
OK. Solved as other issues, by installing additional python package on Ubuntu 20.04LTS (inspired from #449)

sudo apt-get install python3-venv

closed 2021-01-21 by koalabi