platformio/platformio-core-installer #429: Installation Manager

kfine100, 2021-01-07:
How to solve this complaint???

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Linux, 5.4.0-58-generic, x64

  %23 Exception
  ```
  Error: Error: Compatible PlatformIO Core not found.
Reason: PlatformIO installed using another platform `Linux-5.4.0-58-generic-x86_64-with-glibc2.29`. Your platform: Linux-5.4.0-58-generic-x86_64-with-glibc2.10

	at /home/kevin/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275
	at ChildProcess.c (/home/kevin/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:223:5)
	at Pipe.<anonymous> (net.js:664:12)
  ```

ivankravets, 2021-01-21:
Have you resolved this issue?

ivankravets, 2021-01-21:
Please remove `~/.platformio/penv` folder and restart VSCode.

kfine100, 2021-01-24:
Yes, removing penv does make the error go away.

Thanks

kfine100, 2021-01-28:
While the above solutions work they are temporary. The error always comes back after a couple of weeks. I keep my Ubuntu 20.04 up to date and recently did one..........could this be related? Here is the error that is appearing again.

Kevin

Error: Error: Compatible PlatformIO Core not found.
Reason: PlatformIO installed using another platform `Linux-5.4.0-65-generic-x86_64-with-glibc2.10`. Your platform: Linux-5.4.0-65-generic-x86_64-with-glibc2.29

	at /home/kevin/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275
	at ChildProcess.c (/home/kevin/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)...

ivankravets, 2021-01-28:
Where do you see this error?

kfine100, 2021-01-28:
Hi Ivan

When I start VS Code, the following installation window pops up. This one finishes with "PlatformIO IDE installed successfully" but I sometimes instead get the above error.

[image]

ivankravets, 2021-01-28:
This is correct. You do OS updates, it changes system libraries, we need to recreate virtual environemnt.

kfine100, 2021-01-28:
So really there is no error? Looking at the past prints there is an evolution of libraries:

Linux-5.4.0-58-generic-x86_64-with-glibc2.10
Linux-5.4.0-65-generic-x86_64-with-glibc2.10
Linux-5.4.0-65-generic-x86_64-with-glibc2.29

due to Ubuntu updates. The PIO installation is just adapting to the new ones.

I also occasionally get an error about multiple installed platforms. Maybe this is because one of the previous installations sticks around. I suppose I can always clean that out by removing .penv?

Kevin

ivankravets, 2021-01-29:
Thanks, I reopened this issue. We will think of a better solution.

kfine100, 2021-01-29:
You guys are doing a great job. If others responded as quick as you, the world would be a better place!

closed 2021-11-05 by ivankravets