platformio/platformio-core-installer #438: Installation Manager

timsenecal, 2021-01-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.52.1
  PIO IDE: v2.2.1
  System: Linux, 5.4.0-60-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.6/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.6/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/usr/lib/python3/dist-packages/pip/__init__.py", line 29, in <module>
    from pip.utils import get_installed_distributions, get_prog
  File "/usr/lib/python3/dist-packages/pip/utils/__init__.py", line 30, in <module>
    from pip._vendor.retrying import retry
ModuleNotFoundError: No module named 'pip._vendor.retrying'
Error: Could not install PlatformIO Core: Command '['/home/tim/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 0.3.5
Platform: Linux-5.4.0-60-generic-x86_64-with-glibc2.25
Python version: 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/tim/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core

	at /home/tim/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275
	at ChildProcess.c (/home/tim/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

ivankravets, 2021-01-21:
Have you resolved this issue?

timsenecal, 2021-01-21:
no

ivankravets, 2021-01-21:
Could you try to run `pip install -U pip` and restart VSCode?

timsenecal, 2021-01-21:
i have both python2 and python3.  with python2 as my default.
i had to install several packages for python3 as pip3 and not pip.  i just continued to copy paste failures from platformio, replacing pip with pip3

ivankravets, 2021-01-21:
Sorry, something wrong with your machine. Please fix Python 3 installation and `python3-venv` package.

closed 2021-01-21 by ivankravets