platformio/platformio-core-installer #449: Installation Manager

burtonscomputers, 2021-01-20:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.51.1
  PIO IDE: v2.2.1
  System: Linux, 5.4.0-62-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/glenn/.platformio/.cache/tmp/get-platformio-0.3.5.py", line 19, in <module>
    import shutil
  File "/usr/local/lib/python2.7/shutil.py", line 12, in <module>
    import collections
  File "/usr/local/lib/python2.7/collections.py", line 20, in <module>
    from _collections import deque, defaultdict
ImportError: No module named _collections

	at /home/glenn/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879275
	at ChildProcess.c (/home/glenn/.vscode/extensions/platformio.platformio-ide-2.2.1/node_modules/platformio-node-helpers/dist/index.js:1:879169)
	at ChildProcess.emit (events.js:223:5)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:223:5)
	at Pipe.<anonymous> (net.js:664:12)
  ```

ivankravets, 2021-01-21:
Please install Python 3.6+ and restart VSCode.

closed 2021-01-21 by ivankravets

burtonscomputers, 2021-01-21:
Thanks but this was on an Ubuntu system with python 2.7 and several python 3.x installations including 3.9.0. I am not sure how VS Code and PlatformIO stopped recognising that. It was all working fine a week to two back!

Thanks anyway.

ivankravets, 2021-01-21:
It seems VSCode runs from another shell that does not have access to your multiple Python 3. So, please fix your machine.

burtonscomputers, 2021-01-21:
I think this indicates VSCode is seeing all of the versions including the
latest (3.9.1), doesn't it?
[image: image.png]
I'm sorry I do not understand enough about how these multiple versions
co-exist, or even how they got there.
I typically only us "apt" to install and update things on the system.

Glenn Burton
Burtons' Computers <http://burtonscomputers.co.uk/>

ivankravets, 2021-01-22:
You can run the installer manually https://docs.platformio.org/en/latest/core/installation.html#super-quick-mac-linux

```
python3 get-platformio.py
```

burtonscomputers, 2021-01-22:
The manual installation seemed to run without problems. But after I run
VSCode I still get this error.:-
[image: image.png]
If I go to the /home/glenn/.platformio/.cache/tmp directory and run "python
get-platformio-0.3.5.py" I get the same errors. If, instead I run  "python3
get-platformio-0.3.5.py" it seems to run correctly.  I have added
~/.platformio/penv/bin to my path as suggested, logged out and in again.
But I still am getting the error from PlatformIO when I start-up VSCode.

Any help would be appreciated.

Glenn Burton.

ivankravets, 2021-01-22:
It seems that VSCode runs from the shell where there is no `python3`. Please fix your shell configuration. bash, zsh, etc